mne-tools/mne-python #4712: example fails with TypeError on Windows, 32/64 bitness issue

jjnurminen, 2017-10-26:
Trying to run plot_topo_compare_conditions on a Windows machine, it bombs with:

```
...
  File "C:\Users\Vicon123\projects\mne-python\mne\epochs.py", line 829, in _name
    count = np.bincount(self.events[:, 2])

TypeError: Cannot cast array data from dtype('>u4') to dtype('int32') according to the rule 'safe'
```

It looks like events is created as 64-bit unsigned ints, which numpy refuses to cast to 32 bit.
Probably caused by running a 32-bit Python interpreter on a 64-bit machine (I don't have 64-bit Python handy to test with)

agramfort, 2017-10-26:
windows ints are 32 bits even on win64

can it be tested with appveyor?

larsoner, 2017-10-26:
I think we can test 32 bit there, but I doubt that the environment file
will work (although we can try)

agramfort, 2017-10-26:
is it really a win32 issue?

larsoner, 2017-10-26:
In that example we load the events from disk. Maybe that ends up being the problem. @jjnurminen what is the full traceback?

jjnurminen, 2017-10-27:
```
Current compensation grade : 0
288 matching events found
Created an SSP operator (subspace dimension = 3)
4 projection items activated
    Rejecting  epoch based on MAG : [u'MEG 1711']
Traceback (most recent call last):

  File "<ipython-input-2-e18713e46c19>", line 1, in <module>
    runfile('C:/Users/Vicon123/projects/mne-python/examples/visualization/plot_topo_compare_conditions.py', wdir='C:/Users/Vicon123/projects/mne-python/examples/visualization')

  File "C:\Anaconda2\lib\site-packages\spyder\utils\site\sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda2\lib\site-packages\spyder\utils\site\sitecustomize.py", line 87, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/Users/Vicon123/projects/mne-python/examples/visualization/plot_topo_compare_conditions.py", line 58, in <module>
    evokeds = [epochs[name].average() for name in ('left', 'right')]

  File "C:\Users\Vicon123\projects\mne-python\mne\epochs.py", line 751, in average
    return self._compute_mean_or_stderr(picks, 'ave')

  File "C:\Users\Vicon123\projects\mne-python\mne\epochs.py", line 820, in _compute_mean_or_stderr
    kind, self._name)

  File "C:\Users\Vicon123\projects\mne-python\mne\epochs.py", line 828, in _name
    count = np.bincount(self.events[:, 2])

TypeError: Cannot cast array data from dtype('>u4') to dtype('int32') according to the rule 'safe'
```

agramfort, 2017-10-27:
what if you change the dtype of self.events something like

self.events = self.events.astype(int)

or anything that works.

larsoner, 2017-11-08:
I can't replicate on Windows 10. Which Windows version are you on?

jjnurminen, 2017-11-08:
>Which Windows version are you on?

Windows 7. Thanks for the fix, I will try it tomorrow, right now I'm comfortably far from the Windows machine :)

jjnurminen, 2017-11-08:
>Which Windows version are you on?

Windows 7. Thanks for the fix, I will try it tomorrow, right now I'm comfortably far from my Windows machine :)

PR #4744 (2017-11-09): MRG: Fix event dtype for Windows
Closes #4712.

@jjnurminen can you see if this fixes your problem?
closed 2017-11-09 by agramfort