mne-tools/mne-python #4714: Figures from plot_connectivity_circle - content cut off

jhouck, 2017-10-27:
Hi all, 

I'd had this problem with my data and replicated it using the sample data with the plot_mne_inverse_label_connectivity example.  The output includes a black circle floating on a white background; the circle includes the connectivity plot, but the labels are cut off, as are the title and the text of the legend.  
[image]

I don't call a specific backend and don't have an active matplotlibrc, although the output is the same when I do (agg, WXAgg, Qt4Agg, etc.), and if I create the figure with a black facecolor before populating it with plot_connectivity_circle.  Interactivity and show also don't make a difference -- the figure looks fine on screen but saves like the above.  This is on 0.16.dev.0 with matplotlib 1.4.3.

Any suggestions?

larsoner, 2017-10-27:
Try `savefig(..., facecolor=fig.get_facecolor())`. By default matplotlib does not save the facecolor

jhouck, 2017-10-27:
Thanks, that did the trick.  And of course now that you say that I can find examples in my old scripts. 

Mind if I do a small PR to add this to the two plot_connectivity_circle examples? This would reduce the likelihood of me having the same question a year or two from now.

larsoner, 2017-10-27:
Fine by me, but let's leave the `savefig` line commented out so that it doesn't pollute people's directories when they run the examples

larsoner, 2017-11-02:
Reopening as this came up on the listserv again, we should add to the `plot_circle` and example some mention of this problem

jhouck, 2017-11-02:
I can do this right now

jhouck, 2017-11-02:
Derp, one more thing

closed 2018-02-21 by larsoner