mne-tools/mne-python #4715: CNT epoch timestamp access

kk1995, 2017-10-27:
I am new to MNE so it may be that I cannot find the resource. However, I cannot seem to create epoch objects from .cnt (Neuroscan) data alone, when .cnt data can contain timestamps of stimulus presentation.

One possible solution is to make it possible for mne.Epochs to be derived from .cnt files.

agramfort, 2017-10-29:
if your cnt contains continuous recordings (we call Raw in MNE) then you
will need events
that tell you when stimuli were presented. Then you'll be able to have
Epochs.
Now it's possible that our MNE reader does not properly read your events
present in the file.

massich, 2019-03-13:
Once #6047 is merged, this should work also with your files.

```py
import os.path as op
import mne

data_path = mne.datasets.testing.data_path(download=False)
fname = op.join(data_path, 'CNT', 'scan41_short.cnt')

raw = mne.io.read_raw_cnt(fname, montage=None)
events, event_id = mne.events_from_annotations(raw)

epochs = mne.Epochs(raw, events, event_id=event_id)
```

Closing. @kk1995 feel free to re-open if you have any other issues.

closed 2019-03-13 by massich