mne-tools/mne-python #4720: cannot work with .easy files

aitorme, 2017-11-01:
Hi,
I use a mobile EEG from [Neuroelectrics](http://www.neuroelectrics.com/) and their EEG recordings have a file named `.easy` as output. I cannot find any support to read this kind of file in your documentation. I also converted it to .edf with EEGLAB in Matlab, but I got the following error when reading it:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/mne/io/edf/edf.py", line 1235, in read_raw_edf
    exclude=exclude, preload=preload, verbose=verbose)
  File "<string>", line 2, in __init__
  File "/usr/local/lib/python3.6/site-packages/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/mne/io/edf/edf.py", line 164, in __init__
    annotmap, eog, misc, exclude, preload)
  File "/usr/local/lib/python3.6/site-packages/mne/io/edf/edf.py", line 416, in _get_info
    edf_info = _read_gdf_header(fname, stim_channel, exclude)
  File "/usr/local/lib/python3.6/site-packages/mne/io/edf/edf.py", line 710, in _read_gdf_header
    edf_info['number'] = float(version[4:])
ValueError: could not convert string to float: `
Is there a way to work it out directly with `.easy` files? Is there a workaround for this string to float conversion error?
Thanks in advance!

agramfort, 2017-11-01:
try importing the .set file

and use mne.io.read_raw_eeglab

aitorme, 2017-11-01:
there is not `.set` file created by `Neuroelectrics`, only a file called `.easy`. Do you have a way to convert `.easy` into `.set`?

agramfort, 2017-11-01:
I mean save the eeglab file to .set and not to .edf

aitorme, 2017-11-02:
okay, thanks. I guess it's not possible to work with `.easy` files directly, so I will close the ticket since there is a workaround

closed 2017-11-02 by aitorme