mne-tools/mne-python #4724: confliciting logic?

samiabboud, 2017-11-03:
Hey there,

I wanted to see with you if I'm missing something when running a spatio_temporal_cluster_test with tail=-1 and tfce thresholds. 

It seems that with tail=-1, one must use negative start and step tfce values:
https://github.com/mne-tools/mne-python/blob/966e862b95861de47ffe259bd66c86feece524f6/mne/stats/cluster_level.py#L301-L306

When doing so, the check if thresholds are monotonic fails.
https://github.com/mne-tools/mne-python/blob/966e862b95861de47ffe259bd66c86feece524f6/mne/stats/cluster_level.py#L335-L336
Should this check be for absolute values?

larsoner, 2017-11-03:
This does seem like a bug. The check should probably be for `< 0` for the case where `tail=-1` and `> 0` (what's there now) otherwise. PR welcome

cbrnr, 2017-11-08:
Wouldn't it make more sense to require the `start` and `step` values to be always positive? Why should I need to change them just because I want to do a left-tailed test?

cbrnr, 2017-11-08:
On second thought, this might break too much existing code. I've fixed this in #4740.

larsoner, 2017-11-08:
The idea is that it's like saying you want a positive threshold for what are all negative values. It's a bit of an implicit vs explicit distinction.

To be consistent with float arguments to `threshold` we should probably make them negative for now. I see you've done this in #4740 -- thanks.

We could consider allowing positive arguments in both places, and if doing a left (negative value) test, assume that we should negate them, but we can do it separately.

PR #4740 (2017-11-09): [MRG] Fix threshold must be monotonic condition
Fix #4724
closed 2017-11-09 by larsoner