mne-tools/mne-python #4727: Force scalp surface routine skip topology check

ktavabi, 2017-11-04:
Shouldn't `--force` arg allow for skipping topology checks in `mne make_scalp_surfaces`? Presently it's is returning topology error for volume(s) I don't see anything wrong with. 

[image]

> RuntimeError: Surface outer skin is not complete (sum of solid angles = 6.2829 * 4*PI instead).

cc @larsoner

ktavabi, 2017-11-07:
@larsoner Does this sound reasonable to you?

- [ ] Add bool arg to `_surfaces_to_bem` inherited from `mne_make_scalp_surfaces`
- [ ] If `--force` then skip `_check_surface_size` for one or all surfs?

larsoner, 2017-11-07:
I wouldn't skip it but rather turn the error into a warning. If you can share privately the offending surfaces I can take a look

larsoner, 2018-02-21:
This was fixed IIRC

closed 2018-02-21 by larsoner

ktavabi, 2018-02-21:
Yes it was. Thanks.

ckiefer0, 2018-03-28:
I am getting the same error. How can I visualize my data to check for problems (similar to the image provided by @ktavabi)? If the volumes look fine I can just use '--force' to suppress the error, however, if there is a problem I have to obtain a new MR scan?

ktavabi, 2018-03-28:
@ckiefer0 this fix concerned surface topology check. I only use '--force' to skip topology check for problematic outer skin surfaces. I wouldn't recommend skipping any sanity checks for other surfaces. Use the FS FreeView viewer to examine volumes & corresponding surfaces to ensure you have good segmentation and if need be you can use `--force` to skip topology check for the outer skin. In my experience, it is signal dropouts in the neck and face that seem to cause issues. HTH.