mne-tools/mne-python #4731: How do I perform a simple cluster permutation test?

cbrnr, 2017-11-06:
I don't think that I got this right because the results look weird. I want to compute a simple cluster permutation test on EEG data between two groups (control vs. experimental). I've computed a power measure in a specific frequency band for each channel and stored that in two separate arrays `x1` and `x2` both of shape `(31, 64)`. So each group consists of 31 subjects with values measured at 64 channels.

I thought that this would do the job:

```
t, cl, p, h0 = pctest([x1, x2], connectivity=con)
```

where `con, _ = read_ch_connectivity("biosemi64")`.

But the result looks strange, I get one big cluster which looks like this:

[image]

I expected to see a cluster around the dark red values instead.

This is the code I used for plotting:

```
montage = read_montage("biosemi64")
plot_topomap(t.squeeze(), montage.get_pos2d()[:64, :], mask=cl[0])
```

Any comments are greatly appreciated! I don't even know if I'm on the right track or if I'm doing it all wrong (wrong functions - there are so many cluster-related functions in MNE).

larsoner, 2017-11-06:
This could be correct depending on the value you chose for your threshold and the data / noise characteristics of the data.

Can you add a colorbar to the topomap? This could be a correct result (in the sense that the function does what it says it does) if the filled circles all seem to exceed the threshold value used, but you'll have to check the docstring of the `pctest` function (whichever that was) to determine the default (or hopefully using `verbose=True` it will tell you -- if it doesn't we should update it to give the threshold). Everything here is red -- there are no blue / negative values -- so it seems like your statistic is greatly biased toward one condition.

If you changed your threshold value to be more restrictive, you would probably obtain a cluster around the dark red values.

cbrnr, 2017-11-07:
Thanks @larsoner, good to know that my approach looks correct! I forgot to mention that I'm using `from mne.stats import permutation_cluster_test as pctest`.

You're right, the difference is always in one direction, but this is OK.

Some additional questions:

1. The function doesn't tell me the threshold - we should probably write this info to the logger.
2. It seems like the function does not return t-values (as described in the docstring), but statistics as defined by `stat_fun` (which by default is `f_oneway`, so the returned statistics are F-values and not t-values).
3. How do I manually set the threshold? Is there any best practice? I assume this is the reason why TFCE exists...
4. Speaking of TFCE, the documentation could also be improved. It doesn't state how this dict must look like. From the KeyError message, I learned that it must have at least "start" and "step" keys - but I have no idea which values to use for these keys. Playing around a bit with these values, all I managed to come up with are 64 single-electrode clusters for my 64-channel data.
5. What if I have one between and one within factor? Can MNE handle this in a cluster test?

If you want (and agree that it should be done), I could tackle 1 and 2 in a PR. Issues 3 and 4 are docstring improvements, but I can't do it (yet).

larsoner, 2017-11-07:
> The function doesn't tell me the threshold - we should probably write this info to the logger.

Agreed

> It seems like the function does not return t-values (as described in the docstring), but statistics as defined by stat_fun (which by default is f_oneway, so the returned statistics are F-values and not t-values).

Agreed, That's copy-paste error because in the paired code (`*_1samp`) it uses a t-test

> Speaking of TFCE, the documentation could also be improved.

Did you see [this](https://mne-tools.github.io/stable/auto_tutorials/plot_stats_cluster_methods.html)? We should probably add cross links. We should also document the parameter itself better, as you say.

> How do I manually set the threshold? Is there any best practice? I assume this is the reason why TFCE exists...

In general something like the equivalent of `p < 0.05` for your `stat_fun` is a reasonable choice, but it depends on tradeoffs that are data-dependent (and yes `TFCE` is meant to overcome these issues).

> What if I have one between and one within factor? Can MNE handle this in a cluster test?

I *think* this is doable but we need to check how well `statsmodels` supports such models. If it does, you can set up the model there and wrap it in a `partial` or new function def as necessary to match the API MNE wants. If you get it working, it would be great to add another example, probably with toy data with known effects (which I'd advise doing anyway to make sure you know it's working properly before unleashing on real data).

cbrnr, 2017-11-07:
> Agreed, That's copy-paste error because in the paired code (`*_1samp`) it uses a t-test

But the automatically determined threshold uses `scipy.stats.f.ppf` - isn't this only valid if indeed `stat_fun=f_oneway`?

> Did you see this? We should probably add cross links. We should also document the parameter itself better, as you say.

Yes, I saw that at some point but a direct link would certainly be a good idea! But even this example doesn't explain the TFCE threshold dict.

I'll see what I can do about the multi-factor model case, but this sounds like it could take a while 😄 .

cbrnr, 2017-11-07:
Logging is fixed in #4733, I'll wait for the docstring fix until we've resolved the automatic threshold issue.

larsoner, 2017-11-07:
> But the automatically determined threshold uses scipy.stats.f.ppf - isn't this only valid if indeed stat_fun=f_oneway?

Yes, and we should probably document that it should only be left as `None` if the default function `f_oneway` is used.

cbrnr, 2017-11-07:
If you have time please go ahead, you are more familiar with this topic. I'd be happy to review if you want.

cbrnr, 2017-11-08:
Quick checklist of unresolved issues from this thread:

- [ ] Update docstring to reflect that F-values are returned by default
- [ ] Improve TFCE dict argument documentation
- [ ] Automatic threshold works only if `stat_fun` is the default `f_oneway`

cbrnr, 2017-11-08:
OK to close this issue (since my cluster test approach worked) and create new issues?

larsoner, 2017-11-08:
Or just retitle this and put your checklist at the top

larsoner, 2017-11-08:
@cbrnr don't bother, I'm making a PR now

closed 2017-11-08 by larsoner