mne-tools/mne-python #4742: possible BUG: dropping epochs by visual inspection

stfnrpplngr, 2017-11-08:
Found a problem regarding dropping epochs. You can try it with the data of the [read epochs example](https://mne-tools.github.io/dev/auto_examples/io/plot_read_epochs.html#sphx-glr-auto-examples-io-plot-read-epochs-py). 
Plotting epochs and selecting bad epochs by hand results in a reduction of the total number of epochs. This does not work plotting only the epochs corresponding to a certain `event_id`.

Doing it for the whole set:
```
epochs
Out[2]: <Epochs  |  n_events : 55 (all good), tmin : -0.199795213158 (s), tmax : 0.499488032896 (s), baseline : (None, 0), ~17.5 MB, data loaded>

epochs.plot()
Out[3]: <matplotlib.figure.Figure at 0xc8afda0>Dropped 5 epochs
Channels marked as bad: []

epochs
Out[4]: <Epochs  |  n_events : 50 (all good), tmin : -0.199795213158 (s), tmax : 0.499488032896 (s), baseline : (None, 0), ~16.3 MB, data loaded>
```
5 epochs are marked as bad -> the total number of epochs is reduced by five

Plotting only epochs corresponding to an `event_id`:
```
epochs['1']
Out[2]: <Epochs  |  n_events : 55 (all good), tmin : -0.199795213158 (s), tmax : 0.499488032896 (s), baseline : (None, 0), ~17.5 MB, data loaded>

epochs['1'].plot()
Out[3]: <matplotlib.figure.Figure at 0xa250978>Dropped 5 epochs
Channels marked as bad: []

epochs['1']
Out[4]: <Epochs  |  n_events : 55 (all good), tmin : -0.199795213158 (s), tmax : 0.499488032896 (s), baseline : (None, 0), ~17.5 MB, data loaded>
```
5 epochs are marked as bad -> the total number of epochs stays the same

larsoner, 2017-11-08:
Doing `epochs[...]` will make a copy, not a reference to the original object. So to do what you want, you'd need to use `epochs.selection` and do things manually I think.

Is there somewhere in the docs we can make this clearer?

stfnrpplngr, 2017-11-09:
I tried this:
`epochs[epochs['1'].selection].plot()`
But it doesn't work either, same problem. Is there any way to work on a subset of `epochs` instead of plotting all at once?

agramfort, 2017-11-09:
yes it also makes a copy.

what do you mean work? plot? if so we would need to add a param to plot to
only plot a subset of epochs

stfnrpplngr, 2017-11-09:
By work I mean selecting bad epochs.

agramfort, 2017-11-09:
you need to make new epochs per conditions and work individually.
you can concatenate back if you need.

stfnrpplngr, 2017-11-09:
According to your suggestion, this is my approach:
```
epochs_all = list()
counter = 0
for event_id in epochs.event_id:
    epochs_all.append(epochs[str(event_id)])
    epochs_all[counter].plot(picks=picks, scalings='auto', block=True)
epochs_all = mne.concatenate_epochs(epochs_all)
```
It works, but I lose the information about the original timing and indexation.

larsoner, 2017-11-09:
I wouldn't use `concatenate_epochs` (sorry @agramfort). It should be possible to use `epochs.selection` using array manipulation to get the indices back to the original Epochs instance. I see you tried above but it seems like it must be possible. If you're sick of trying I can give it a shot.

stfnrpplngr, 2017-11-09:
Currently, I have no further ideas how to apply the `epochs.selection` except as indices ...

larsoner, 2017-11-09:
Argh yes `selection` gets restricted. Fortunately we can use event numbers, though. See if this makes sense:
```
import numpy as np
import mne
raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() +
                          '/MEG/sample/sample_audvis_raw.fif')
events = mne.find_events(raw, 'STI 014')
epochs = mne.Epochs(raw, events, event_id=dict(foo=1, bar=2), preload=True)
epochs.drop([1])
epochs_foo = epochs['foo']
# we can get indices back to the original using np.searchsorted
# based on the epoch sample numbers (because they're in order):
orig_idx = np.searchsorted(epochs.events[:, 0], epochs_foo.events[:, 0])
assert (epochs.events[orig_idx, 2] == 1).all()  # sanity check
# let's drop some in our copy
epochs_foo.drop([0, 1, 2])
# figure out which ones are still there
orig_idx_good = np.searchsorted(epochs.events[:, 0], epochs_foo.events[:, 0])
assert len(orig_idx_good) == len(orig_idx) - 3  # we removed 3
# figure out which ones we removed (in the original)
orig_idx_bad = np.setdiff1d(orig_idx, orig_idx_good)
assert len(orig_idx_bad) == 3
epochs.drop(orig_idx_bad)
# Did we drop the correct ones?
assert np.array_equal(epochs['foo'].events, epochs_foo.events)
```

stfnrpplngr, 2017-11-13:
Thanks for your help. I think it works that way, with only slight modifications:

```
        for event_id in epochs.event_id:
            epochs_event = epochs[event_id]
            # we can get indices back to the original using np.searchsorted
            # based on the epoch sample numbers (because they're in order):
            orig_idx = np.searchsorted(epochs.events[:, 0], epochs_event.events[:, 0])
            assert (epochs.events[orig_idx, 2] == int(event_id)).all()  # sanity check
            # let's drop some in our copy
            epochs_event.plot(picks=picks, scalings='auto',
                              block=True) # epochs_foo.drop([0, 1, 2])
            # figure out which ones are still there
            orig_idx_good = np.searchsorted(epochs.events[:, 0], epochs_event.events[:, 0])
            idx_bads = [idx for idx in range(len(epochs_event.drop_log)) \
                        if epochs_event.drop_log[idx] == ['USER']]
            n_bads = len(idx_bads)
            assert len(orig_idx_good) == len(orig_idx) - n_bads  # we removed 3
            # figure out which ones we removed (in the original)
            orig_idx_bad = np.setdiff1d(orig_idx, orig_idx_good)
            assert len(orig_idx_bad) == n_bads
            epochs.drop(orig_idx_bad, reason='USER_'+event_id)
            # Did we drop the correct ones?
            assert np.array_equal(epochs[event_id].events, epochs_event.events) 
```

Nevertheless, is it conceivable that in future they will be a param to plot only a subset of epochs without copying? Furthermore, it would be nice if marked epochs would not be deleted immediately. Instead they could get annotated like it is done for segments in `raw`.

larsoner, 2018-04-04:
Closing since the main problem can by solved by a NumPy-based workflow. @stfnrpplngr if you are still interested in the other features you mention in your last post, feel free to open new issues for discussion

closed 2018-04-04 by larsoner