mne-tools/mne-python #4746: Move evoked topo plot to plot_compare_evokeds

jona-sassenhagen, 2017-11-13:
Evoked topoplots should make use of plot_compare_evokeds.
I.e., something like this

```
%matplotlib inline
from itertools import combinations

from scipy import stats

import matplotlib.pyplot as plt
import numpy as np

import mne
from mne.datasets import testing
from mne.channels.layout import _find_topomap_coords
from mne.stats.cluster_level import spatio_temporal_cluster_1samp_test as tst
from mne.defaults import DEFAULTS

data_path = testing.data_path()
fname = data_path + "/EEGLAB/test_raw.set"
montage = data_path + "/EEGLAB/test_chans.locs"

event_id = {"rt":1, "square":2}
eog = {"FPz", "EOG1", "EOG2"}
raw = mne.io.eeglab.read_raw_eeglab(fname, preload=True, eog=eog,
                                    montage=montage, event_id=event_id
                                   )#.filter(None, 20)
events = mne.find_events(raw)
picks = mne.pick_types(raw.info, eeg=True)
epochs = mne.Epochs(raw, events,  event_id,#{"square":2},
                    tmax=.7, picks=picks)
evoked = epochs["square"].average()

from mne.channels.layout import find_layout
layout = find_layout(evoked.info)
pos = layout.pos.copy()

f = plt.figure()
f.set_size_inches((10, 10))

evokeds = {cond:list(epochs[cond].iter_evoked()) for cond in event_id}
ylims = (evoked.data.min() * DEFAULTS["scalings"]["eeg"],
         evoked.data.max() * DEFAULTS["scalings"]["eeg"])
ylims = (-30, 40)
ymax = np.min(np.abs(np.array(ylims)))
for pick, (pos_, ch_name) in enumerate(zip(pos, evoked.ch_names)):
    ax = plt.axes(pos_)
    mne.viz.plot_compare_evokeds(evokeds, picks=pick, axes=ax,
                         ylim=dict(eeg=ylims),
                         show=False,
                         show_sensors=False,
                        show_legend=False,
                         title='');
    ax.set_xticklabels(())
    ax.set_ylabel('')
    ax.set_xlabel('')
    ax.set_yticks((-ymax, ymax))
    ax.spines["left"].set_bounds(-ymax, ymax)
    ax.set_ylim(ylims)
    ax.set_yticklabels('')
    ax.text(-.1, 1, ch_name, transform=ax.transAxes)

ax_l = plt.axes([0, 0] + list(pos[0, 2:]))
mne.viz.plot_compare_evokeds(evokeds, ylim=dict(eeg=ylims), title='', show_sensors=False,
                     picks=0, axes=ax_l, ci=None,
                             show=False)

ax_l.set_yticks((-ymax, ymax))
ax_l.spines["left"].set_bounds(-ymax, ymax)
ax_l.set_ylim(ylims)
ax_l.lines.clear()
ax_l.patches.clear()
```

[image]

mmagnuski, 2017-11-14:
Looks beautiful!

jona-sassenhagen, 2017-11-14:
I think so too :)

I'm not perfectly sure how to integrate it though. (Also I would never use this kind of plot myself.)

jona-sassenhagen, 2017-11-14:
But I will try getting around to it eventually.

jona-sassenhagen, 2018-05-01:
@sappelhoff please see here. I'm still working on the code, I can just give you my code and you could take over? It's not entirely trivial though.

sappelhoff, 2018-05-01:
Thanks for the link Jona. I don't know if my taking over would speed up the process but I could certainly give it a try :-)

jona-sassenhagen, 2018-05-02:
Sure, go for it. See this gist:

https://nbviewer.jupyter.org/gist/jona-sassenhagen/e54c80e16fab0c5b635f0c08bae6bef3

for my current experiments.

sappelhoff, 2018-05-03:
What are the steps to be done? It's looking really nice already imo ...

jona-sassenhagen, 2018-05-03:
The problem is making it look nice for multiple layouts and contrasts. i.e., check it on various data ...

Then it needs to become MNE code. Hopefully it can just become the backend for the current API, i.e., swap it in for the code currently in `mne.viz.plot_evoked_topo`.

jona-sassenhagen, 2018-05-31:
@sappelhoff would you be able to implement this eventually?

sappelhoff, 2018-05-31:
@jona-sassenhagen ... let's say it's on my list. But if somebody else wants to do it, I wouldn't be sad ;-) 

There is also still #5186 to do for me (although we should maybe re-decide whether it's worth it)

sappelhoff, 2018-08-10:
@jona-sassenhagen I think I won't get to it after all, sorry.

jona-sassenhagen, 2018-09-23:
@agramfort I would expand the API of plot_evoked_topo to be roughly in alignment with plot_compare_evokeds.

E.g., I would allow dict input (`{"condition": list_of_evoekds}` etc), and a few of the viz things.

Makes sense?

agramfort, 2018-09-23:
no strong feeling. If you can do this with a small refactoring sure.

PR #6211 (2019-04-27): MRG +1 plot_compare_evokeds topo functionality
Another attempt. It's not really hard, but it's my 3rd attempt so ...

mne.viz.plot_compare_evokeds gains the ability to plot to a topographical layout.
This will be slow for MEG data, but is often used for ERP papers.

Not ready for reviews yet.

Closes #4746
Closes #5746
Closes #6166
Closes #6039

Redo of #5568
closed 2019-04-27 by agramfort