mne-tools/mne-python #4748: SlidingEstimator(n_jobs=>1) nested in cross_val_multiscore gets stuck

jona-sassenhagen, 2017-11-14:
As the title says - on my Mac, in the Jupiter notebook, when I'm using SlidingEstimator with n_jobs > 1 nested inside cross_val_multiscore (with any n_jobs), the thing simply hangs - no processor utilisation, not results. Anyone familiar with the problem? Should I make a reproducible example?

@kingjr

agramfort, 2017-11-14:
make sure you use mkl then it should work. It's joblib / mac blas lapack
issue

jona-sassenhagen, 2017-11-14:
Hm. 

```
numpy.show_config()
blas_mkl_info:
    libraries = ['mkl_rt', 'pthread']
    library_dirs = ['/Users/jona/miniconda3/lib']
    define_macros = [('SCIPY_MKL_H', None), ('HAVE_CBLAS', None)]
    include_dirs = ['/Users/jona/miniconda3/include']
blas_opt_info:
    libraries = ['mkl_rt', 'pthread']
    library_dirs = ['/Users/jona/miniconda3/lib']
    define_macros = [('SCIPY_MKL_H', None), ('HAVE_CBLAS', None)]
    include_dirs = ['/Users/jona/miniconda3/include']
lapack_mkl_info:
    libraries = ['mkl_rt', 'pthread']
    library_dirs = ['/Users/jona/miniconda3/lib']
    define_macros = [('SCIPY_MKL_H', None), ('HAVE_CBLAS', None)]
    include_dirs = ['/Users/jona/miniconda3/include']
lapack_opt_info:
    libraries = ['mkl_rt', 'pthread']
    library_dirs = ['/Users/jona/miniconda3/lib']
    define_macros = [('SCIPY_MKL_H', None), ('HAVE_CBLAS', None)]
    include_dirs = ['/Users/jona/miniconda3/include']
```

Might it be related to my sklearn being self compiled?

Do you off-hand know the joblib issue #?

agramfort, 2017-11-15:
see maybe:

https://github.com/numpy/numpy/issues/5752
https://github.com/scikit-learn/scikit-learn/issues/636

closed 2017-11-15 by jona-sassenhagen