mne-tools/mne-python #4750: Loading EDF files fails

cbrnr, 2017-11-14:
The last commit that affected `mne/io/edf/edf.py` (https://github.com/mne-tools/mne-python/commit/9ac6547284bbd73b4c18d7d7031042585388005c#diff-825d45683d0be02fad23614151867e02) broke it. When I load a BDF file like this, I get an error:

```
raw = mne.io.read_raw_edf("test.bdf")
```

```
Traceback (most recent call last):
  File "/Users/clemens/Library/Preferences/PyCharmCE2017.2/scratches/scratch_4.py", line 4, in <module>
    raw = mne.io.read_raw_edf("/Volumes/Begabungsforschung/Analysis/Data/EEG/adds/vp1/vp1_exp_eeg.bdf")
  File "/Users/clemens/Repositories/adds/mne/io/edf/edf.py", line 1229, in read_raw_edf
    exclude=exclude, preload=preload, verbose=verbose)
  File "<string>", line 2, in __init__
  File "/Users/clemens/Repositories/adds/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/Users/clemens/Repositories/adds/mne/io/edf/edf.py", line 158, in __init__
    annotmap, eog, misc, exclude, preload)
  File "/Users/clemens/Repositories/adds/mne/io/edf/edf.py", line 564, in _get_info
    tal_channel is None and n_samps[stim_channel] != int(max_samp):
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

However, if I set `stim_channel='auto'` explicitly, it works. This is strange since this is also the default value.

larsoner, 2017-11-14:
So this is almost certainly related to the same commit/PR as #4710, but is now a new breakage mode?

It looks like these are documented clearly enough that we should be able to fix it. I can try to find time to fix it in the next week or so unless someone else beats me to it. We'll need to fix and backport to 0.15.

cbrnr, 2017-11-14:
> So this is almost certainly related to the same commit/PR as #4710, but is now a new breakage mode?

This error was introduced by 9ac6547. I don't understand why it occurs yet, because explicitly setting `stim_channel='auto'` should be the same as using the default - but it's not.

No need to backport, this commit was after 0.15.

larsoner, 2017-11-14:
It is different in the reader `read_raw_edf`:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L1154

Than in the class `RawEDF`:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L153

Can you make a quick PR to fix it? If not I can get to it later this week.

cbrnr, 2017-11-14:
Alright, this should be fixed in #4751

PR #4751 (2017-11-14): Set correct default stim_channel arg
Fix #4750
closed 2017-11-14 by larsoner