mne-tools/mne-python #4754: Float index in coregistration gui

mmagnuski, 2017-11-14:
@christianbrodbeck 
I have a problem with the source alignment example - I get the following error when trying to open the GUI:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-12-8dc6699564c7> in <module>()
----> 1 mne.gui.coregistration(subject='sample', subjects_dir=subjects_dir)

c:\proj\src\mne-python\mne\gui\__init__.py in coregistration(tabbed, split, scene_width, inst, subject, subjects_dir, guess_mri_subject, scene_height, head_opacity, head_high_res, trans, scrollable, verbose)

c:\proj\src\mne-python\mne\utils.py in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

c:\proj\src\mne-python\mne\gui\__init__.py in coregistration(tabbed, split, scene_width, inst, subject, subjects_dir, guess_mri_subject, scene_height, head_opacity, head_high_res, trans, scrollable, verbose)
    127     frame = CoregFrame(inst, subject, subjects_dir, guess_mri_subject,
    128                        head_opacity, head_high_res, trans, config)
--> 129     return _initialize_gui(frame, view)
    130 
    131 

c:\proj\src\mne-python\mne\gui\__init__.py in _initialize_gui(frame, view)
     16         return frame.edit_traits(view=view), frame
     17     else:
---> 18         frame.configure_traits(view=view)
     19         return frame
     20 

C:\Anaconda3\envs\pyqt4\lib\site-packages\traits\has_traits.py in configure_traits(self, filename, view, kind, edit, context, handler, id, scrollable, **args)
   2154                 context = self
   2155             rc = toolkit().view_application( context, self.trait_view( view ),
-> 2156                                            kind, handler, id, scrollable, args )
   2157             if rc and (filename is not None):
   2158                 fd = None

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\toolkit.py in view_application(self, context, view, kind, handler, id, scrollable, args)
    211         from . import view_application
    212         return view_application.view_application( context, view, kind, handler,
--> 213                                                   id, scrollable, args )
    214 
    215     #---------------------------------------------------------------------------

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\view_application.py in view_application(context, view, kind, handler, id, scrollable, args)
     88                   id         = id,
     89                   scrollable = scrollable,
---> 90                   args       = args )
     91 
     92     # If the UI has not been closed yet, we need to keep a reference to

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\view.py in ui(self, context, parent, kind, view_elements, handler, id, scrollable, args)
    431             kind = self.kind
    432 
--> 433         ui.ui( parent, kind )
    434 
    435         return ui

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\ui.py in ui(self, parent, kind)
    220                                    dispatch = 'ui' )
    221         self.rebuild = getattr( toolkit(), 'ui_' + kind )
--> 222         self.rebuild( self, parent )
    223 
    224     #---------------------------------------------------------------------------

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\toolkit.py in ui_live(self, ui, parent)
    153         """
    154         from . import ui_live
--> 155         ui_live.ui_live( ui, parent )
    156 
    157     def ui_modal ( self, ui, parent ):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_live.py in ui_live(ui, parent)
     37     """Creates a live, non-modal PyQt user interface for a specified UI object.
     38     """
---> 39     _ui_dialog(ui, parent, BaseDialog.NONMODAL)
     40 
     41 def ui_livemodal(ui, parent):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_live.py in _ui_dialog(ui, parent, style)
     57         ui.owner = _LiveWindow()
     58 
---> 59     BaseDialog.display_ui(ui, parent, style)
     60 
     61 

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_base.py in display_ui(ui, parent, style)
    480         """Display the UI."""
    481 
--> 482         ui.owner.init(ui, parent, style)
    483         ui.control = ui.owner.control
    484         ui.control._parent = parent

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_live.py in init(self, ui, parent, style)
    175             bbox = None
    176 
--> 177         self.add_contents(panel(ui), bbox)
    178 
    179     def close(self, rc=True):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in panel(ui)
    253         panel = None
    254     if nr_groups == 1:
--> 255         panel = _GroupPanel(content[0], ui).control
    256     elif nr_groups > 1:
    257         panel = QtGui.QTabWidget()

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in __init__(self, group, ui, suppress_label)
    543             self._setup_editor(group, editor)
    544 
--> 545             self._add_splitter_items(content, splitter)
    546 
    547         elif group.layout in ('tabbed', 'fold'):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in _add_splitter_items(self, content, splitter)
    621             # Get a panel for the Item or Group.
    622             if isinstance(item, Group):
--> 623                 panel = _GroupPanel(item, self.ui, suppress_label=True).control
    624             else:
    625                 panel = self._add_items([item])

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in __init__(self, group, ui, suppress_label)
    588 
    589             if isinstance(content[0], Group):
--> 590                 layout = self._add_groups(content, inner)
    591             else:
    592                 layout = self._add_items(content, inner)

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in _add_groups(self, content, outer)
    667         # Process each group.
    668         for subgroup in content:
--> 669             panel = _GroupPanel(subgroup, self.ui).control
    670 
    671             if isinstance(panel, QtGui.QWidget):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in __init__(self, group, ui, suppress_label)
    588 
    589             if isinstance(content[0], Group):
--> 590                 layout = self._add_groups(content, inner)
    591             else:
    592                 layout = self._add_items(content, inner)

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in _add_groups(self, content, outer)
    667         # Process each group.
    668         for subgroup in content:
--> 669             panel = _GroupPanel(subgroup, self.ui).control
    670 
    671             if isinstance(panel, QtGui.QWidget):

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in __init__(self, group, ui, suppress_label)
    590                 layout = self._add_groups(content, inner)
    591             else:
--> 592                 layout = self._add_items(content, inner)
    593             layout.setAlignment(QtCore.Qt.AlignLeft | QtCore.Qt.AlignTop)
    594 

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\ui_panel.py in _add_items(self, content, outer)
    858             # that it is not None (otherwise the main TraitsUI code can change
    859             # the "kind" of the created UI object).
--> 860             editor.prepare(inner)
    861             control = editor.control
    862 

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\editor.py in prepare(self, parent)
    154             self.context_object.on_trait_change( self._update_editor, name,
    155                                                  dispatch = 'ui' )
--> 156         self.init( parent )
    157         self._sync_values()
    158         self.update_editor()

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\enum_editor.py in init(self, parent)
    367                                self.update_object)
    368 
--> 369         self.rebuild_editor()
    370 
    371     #---------------------------------------------------------------------------

C:\Anaconda3\envs\pyqt4\lib\site-packages\traitsui\qt4\enum_editor.py in rebuild_editor(self)
    431             for j in range( cols ):
    432                 if n > 0:
--> 433                     name = names[index]
    434                     rb = self.create_button(name)
    435                     rb.value = mapping[name]

TypeError: list indices must be integers or slices, not float
```
Seems that `index` is a float, but I don't know how it gets to that point.

christianbrodbeck, 2017-11-15:
Looks like this happens before even starting the GUI? I can't reproduce it with `conda` install of `mne` (on virtual machine Windows 10) so maybe it's due to a specific `traits` version?.

agramfort, 2017-11-15:
yes I've seen this in the past. Never found a solution. With my new conda
env I don't have this pb anymore

larsoner, 2017-11-15:
Yes you need a newer `traitsui`, consider updating to either conda env we provide (environment.yml, environment2.yml)

https://github.com/enthought/traitsui/pull/311

closed 2017-11-15 by larsoner

christianbrodbeck, 2017-11-15:
@larsoner do you know which version? We should require it in the conda recipe.

larsoner, 2017-11-15:
IIRC it's the currently released version. Hence why we pin `git` commits while waiting for them to cut a release.

mmagnuski, 2017-11-15:
Cool, thanks for help!

christianbrodbeck, 2017-11-16:
@mmagnuski what was the `traitsui` version that failed for you?

mmagnuski, 2017-11-17:
The one failing in my base environment:
```
traitsui                  5.1.0                    py35_0    menpo
```
the one I updated to (along with the whole environment2.yml) is `5.2.0.dev296`.

larsoner, 2017-11-17:
Ahh right, I think you need to be on Python3 for the error to show up, and it does with 5.1.0, so we do need to use some unreleased dev version

christianbrodbeck, 2017-11-17:
Hm yeah then we should update the conda-forge recipe once a new version is released...