mne-tools/mne-python #4759: EGI file import

LizetteH, 2017-11-15:
Hi, 

First of all, thank you very much for the EGI file import, it saves us a lot of time!

We have been using the function mne.io.read_raw_egi on our EGI data (version5.3) but encountered several issues:
- There seems to be a trigger exclusion problem, where the first trigger gets excluded. This happens about 50% of the time, it is always the first trigger in the stimulation protocol, but never one of the manually set triggers in the EGI software (that happen before). I set both the include and exclude criteria to None, and all my triggers happen more than once. I have tried to set include to a list of all triggers, this did not help. I have to admit that I am having problems understanding the documentation on the inclusion/exclusion, maybe I am missing something? 

- There is one subject for which we had to pause the recording within the EGI software. The import of this data throws an indexError (out of bounds) as shown in the error message attached. 

[errorMessage_EGIdatapaused.txt](https://github.com/mne-tools/mne-python/files/1474927/errorMessage_EGIdatapaused.txt)

Please let me know if you'd like me to sent you the data with the pause (700mb)

I hope you can help, thank you!

agramfort, 2017-11-17:
@LizetteH you will need to look at the code to figure out what is happening. I don't know if @jaeilepp will find the time to look. It can be useful to share the data so someone can replicate and maybe help.

fraimondo, 2018-03-27:
Can confirm, if you pause the recording, you get this error.

Inside the MFF file you have two epochs.

I don't have time to create the test file and address this issue right now. But if someone does create a 10s file with a pause in between, I can take a look.