mne-tools/mne-python #4760: [DISCUSS] Fix EDF stim channel behavior

cbrnr, 2017-11-16:
The way events from EDF/BDF/GDF are parsed produces unexpected results (#4340, #4710, #4756, #4757). The main issues revolve around the `stim_channel` argument. The new default value changed from `-1` (meaning the last channel in the data is a stim channel) to `'auto'` (meaning if annotations exist then an artificial stim channel is generated and appended to the data, otherwise no stim channel is added). There could be some problems with this.

I'll use `pyedflib` for the ground truth.

```
from mne.io import read_raw_edf
import pyedflib as edf
import pandas as pd
```

### BDF file without EDF annotations
https://www.biosemi.com/download/BDFtestfiles.zip
```
f = edf.EdfReader("Newtest17-256.bdf")
status = f.readSignal(16, 0)
print("Signals: {}".format(f.signals_in_file))
print(", ".join(f.getSignalLabels()))
print("Annotations: {}".format("yes" if f.read_annotation() else "no"))
print("Status channel:\n", pd.Series(status).value_counts(), sep="")
f._close()
```

```
raw = read_raw_edf("Newtest17-256.bdf", stim_channel="auto", preload=True)
print(raw.info.__repr__().split("\n")[4])
print(", ".join(raw.info["ch_names"]))
print("Status channel:\n", pd.Series(raw._data[-1, :]).value_counts(), sep="")
```

All 17 channels are parsed as EEG channels. All labels are correct. The contents of the "Status" channel is also correct.

With the previous default of `stim_channel=-1`, the last channel would (correctly) be a STIM channel. However, it would change its name from "Status" to "STI 014". Furthermore, the values of this channel would also be masked so that they are capped at 8 bits:

```
raw = read_raw_edf("Newtest17-256.bdf", stim_channel=-1, preload=True)
print(raw.info.__repr__().split("\n")[4])
print(", ".join(raw.info["ch_names"]))
print("Status channel:\n", pd.Series(raw._data[-1, :]).value_counts(), sep="")
```

Finally, `stim_channel=None` is identical to `stim_channel="auto"` in this case (because there are no EDF annotations).

### EDF file with annotations
https://www.physionet.org/pn4/eegmmidb/S001/S001R04.edf
```
f = edf.EdfReader("S001R04.edf")
status = f.readAnnotations()
print("Signals: {}".format(f.signals_in_file))
print(", ".join(f.getSignalLabels()))
print("Annotations: {}".format("yes" if f.read_annotation() else "no"))
print("Status channel:\n", status, sep="")
f._close()
```

There are 64 EEG channels and annotations containing 30 events, the first event occurring at t = 0s (the first sample).

```
raw = read_raw_edf("S001R04.edf", stim_channel="auto", preload=True)
print(raw.info.__repr__().split("\n")[4])
print(", ".join(raw.info["ch_names"]))
```

This correctly loads all 64 EEG channels and adds a MISC channel called "STI 014". There are two issues: (1) `mne.find_event` can't parse this generated MISC channel because there's an initial event at sample 0 (but we're working on that in #4756), and (2) the generated channel should be STIM instead of MISC.

```
raw = read_raw_edf("S001R04.edf", stim_channel=-1, preload=True)
print(raw.info.__repr__().split("\n")[4])
print(", ".join(raw.info["ch_names"]))
```

Interestingly, this behaves exactly the same as `stim_channel="auto"`. I don't think this is intended, because by saying `stim_channel=-1` I want the last signal in the file to be a stim channel. Apparently, MNE is always converting EDF annotations into a stim channel first, which is then the last channel.

```
raw = read_raw_edf("S001R04.edf", stim_channel=None, preload=True)
print(raw.info.__repr__().split("\n")[4])
print(", ".join(raw.info["ch_names"]))
```

Now this also results in 64 EEG and 1 MISC channel, but this time the name of the MISC channel is `EDF Annotations-1`, which generates a warning that it is too long and it is truncated to 15 characters. I'd expect `stim_channel=None` to not add any artificial channel.

Now the more I dig into it the more I understand some decisions. But 10 hours ago, most of that didn't make any sense. I believe that it should make sense on first use, otherwise (new) users will be frustrated. Some things that could be changed to resolve most issues:

- `mne.find_event` needs to support initial events (in progress #4756)
- EDF annotations should *only*  be parsed if `stim_channel='auto'`; specifically, if this argument is a channel number (e.g. `-1`) or `None`, EDF annotations should not be used for automatic stim channel generation (but annotations should still be accessible with `raw.find_edf_events`).
- I'm not sure if `stim_channel='auto'` should change the values of the stim channel. This seems to be a problem in BDF files (24bit), which currently get masked to 8bit values.
- The automatically generated stim channel should be STIM and not MISC
- What happens if EDF annotations contain invalid time points (see e.g. #4340, the last annotation is outside the time range of the file)? I think we should issue a warning and discard invalid events in the auto-generated stim channel (we don't touch the annotation list in `raw.find_edf_events` though).
- Finally, in combination with the `montage` argument, `read_raw_edf` now produces an error with `stim_channel='auto'` (see #4757).

Looking forward to discussing these things with you guys.

larsoner, 2017-11-16:
> EDF annotations should only be parsed if stim_channel='auto'; specifically, if this argument is a channel number (e.g. -1) or None, EDF annotations should not be used for automatic stim channel generation (but annotations should still be accessible with raw.find_edf_events).

Agreed

> I'm not sure if stim_channel='auto' should change the values of the stim channel. This seems to be a problem in BDF files (24bit), which currently get masked to 8bit values.

Maybe we need a `masking=True | False` argument. In retrospect I'm not sure if we should really automatically mask at all, but now that we've done it, we should probably leave it as the default behavior and give a way to disable it.

> The automatically generated stim channel should be STIM and not MISC

Agreed

> What happens if EDF annotations contain invalid time points (see e.g. #4340, the last annotation is outside the time range of the file)? I think we should issue a warning and discard invalid events [...]

Sounds good to me, except that I suspect it is expected / normal behavior for the special "STOP" event to be at sample number N. We should either read more about that to figure out if it's true, check more files with annotations, or just assume that it is special (what I'd do) and add a conditional that if the `event_type == 'STOP'` and the sample number is N, we don't warn about that specific one.

> Finally, in combination with the montage argument, read_raw_edf now produces an error with stim_channel='auto'

This must be a bug with the montage code, or how it's used. Ideally these `montage` arguments wouldn't exist since they're redundant with `set_montage` functionality. We should probably refactor the code such that just before returning from `__init__`, all `montage` stuff is handled by a call to `set_montage`. That should pass, and if it doesn't, it means some channel types aren't handled properly in either `__init__` or `set_montage`.

@teonbrooks I think you worked on a lot of this stuff, any thoughts on these?

teonbrooks, 2017-11-16:
> Now the more I dig into it the more I understand some decisions. But 10 hours ago, most of that didn't make any sense. I believe that it should make sense on first use, otherwise (new) users will be [...]

cool, I'm following along there now. it seems this is needed for systems that generate an event at the start of the recording.

>     EDF annotations should only be parsed if stim_channel='auto'; specifically, if this argument is a channel number (e.g. -1) or None, EDF annotations should not be used for automatic stim channel generation (but annotations should still be accessible with raw.find_edf_events).

I think this is fair assuming you mean parsed as a stim channel. do we have a way to convert the annotations to a stim channel if it is not done at the initialization?

>     I'm not sure if stim_channel='auto' should change the values of the stim channel. This seems to be a problem in BDF files (24bit), which currently get masked to 8bit values.

how is it changing the stim channel value? I think the stim channel supports values to 16 bits, but I could be incorrect

>     The automatically generated stim channel should be STIM and not MISC

this seems like a bug, agreed this should be a stim not a misc.

> What happens if EDF annotations contain invalid time points (see e.g. #4340, the last annotation is outside the time range of the file)? I think we should issue a warning and discard invalid events [...]

warnings sound like the way to go and discard of invalid events.

>     Finally, in combination with the montage argument, read_raw_edf now produces an error with stim_channel='auto' (see #4757).

I will look into this. I think this is a matter of the stim channel creation and the montage not filtering on channel types.

cbrnr, 2017-11-17:
I'm pretty sure there are no special event types in EDF:
http://edfplus.info/specs/edfplus.html#edfplusannotations

cbrnr, 2018-02-07:
#4888 fixes loading the two example files in the top post. We'll have to dig through all related issues/PRs to find out if there's anything that's not been fixed.

cbrnr, 2018-03-15:
Last remaining issue fixed in #5019.

closed 2018-03-15 by cbrnr