mne-tools/mne-python #4762: bug in evoked.plot_topomap/plot_joint with restricted channels.

bloyl, 2017-11-16:
I sometimes only use right/left channels for some processing and have noticed that there is some spatial scaling the happens in plot_topomap and plot_joint that make makes the half field data seem to cover the entire head.

Below is a screen shot of the evoked.plot_topomap results and a full working example showing the issue is at https://gist.github.com/bloyl/2f067bcf5f84144e090d45fa01836ced

interestingly the channel map (the upper right corner of plot_joint) has output more inline with what I expected.

[image]

larsoner, 2017-11-16:
I think this is a dup of #3987, can you check?

bloyl, 2017-11-16:
Yes it does seem to be the same underlying issue.

larsoner, 2017-11-16:
Thanks, hopefully someone can find time to fix it soon, I agree it's confusing

closed 2017-11-16 by larsoner