mne-tools/mne-python #4764: find events

hendriksantosa, 2017-11-17:
Hi all, 

I am trying to see the ERP response for my MEG data. 
Anyone can help me with the mne.find_events(raw)? 
I could not find the events using that function. But, I can see the stim-mark using raw.plot(block=True)

Thank you!

Hendrik

agramfort, 2017-11-17:
can you try to specify the name of the stim_channel parameter ?

https://martinos.org/mne/stable/generated/mne.find_events.html

hendriksantosa, 2017-11-17:
I can see the stim-mark now by specify the name of the stim-channel. 
However, my stim-mark was flip. I can get the (shifting) ERP response. Is there any function to reverse the stim-mark? (some kind like multiply by -1)
Thank you

agramfort, 2017-11-17:
you can preload and do the -1* on raw._data attribute (for right row)

jjnurminen, 2017-11-27:
If you're on Elekta/Neuromag system and have defined the averaging categories in the acquisition software, you might benefit from the AcqParserFIF class. See example below:

https://mne-tools.github.io/dev/auto_examples/io/plot_elekta_epochs.html

cbrnr, 2017-11-29:
@hendriksantosa can you share your data so we can have a look? Sometimes, you need to tune some arguments of `mne.find_events`, e.g. `uint_cast`, `shortest_event`, `min_duration`, and `consecutive`.