mne-tools/mne-python #4766: All channels zero after ICA. Bug?

bdyetton, 2017-11-21:
Hi MNE team,
After selecting ICA channels associated with eye movement on the ```plot_sources``` dialog, I apply ICA but all the channels then go to zero. 

Here are the ICA sources:
[image]

If I select ICA009, and then apply the ICA then i get:
[image]

The strange thing is this happens even when I select no components. Why does this occur? Could it be a bug? This does not happen with all eeg records but seems to always happen if the large noisy ICA component is found.

agramfort, 2017-11-21:
hum maybe a scaling issue. Can you share a code snippet to replicate?

bdyetton, 2017-11-21:
Wow, that took some investigating, but I narrowed it down to the ```eeg.set_eeg_reference(projection=False)``` call. Remove that, and all is well. I believe this is because the average reference has already been applied, and then I was accidentally applying it again.

File here:
[test_ica_raw.zip](https://github.com/mne-tools/mne-python/files/1492722/test_ica_raw.zip)

 
Code to reproduce:
`import mne
def ica_error_mintest():
    eeg = mne.io.read_raw_fif('test_ica_raw.fif')
    eeg.load_data()
    eeg.set_eeg_reference(projection=False)
    ica = mne.preprocessing.ICA()
    ica = ica.fit(eeg, reject=dict(eeg=40e-5))
    ica.plot_sources(eeg)
    ica.apply(eeg)
    plt.show()
    eeg.plot()
    plt.show()`

bdyetton, 2017-11-22:
Given this is caused by an error on my end, im closing this.

agramfort, 2017-11-23:
it should not be a problem to apply avg ref multiple times. I am reopening until someone can explain what happened

cbrnr, 2017-11-29:
Actually you can only apply an average reference multiple times if the original reference stays in the data. The reference signal is not in among the 12 channels I assume, so applying another average reference further reduces the rank of the data. I do get nice results when I fit ICA on the double average referenced data like this:

```
ica = mne.preprocessing.ICA(max_pca_components=11)
ica.fit(eeg)
```

agramfort, 2017-11-29:
ok so I guess there is no bug.

cbrnr, 2017-11-29:
IMO yes, there's no bug.

closed 2018-04-04 by larsoner