mne-tools/mne-python #4768: BUG in ICA function find_bads_ecg

jdammers, 2017-11-21:
@agramfort  @dengemann 
1.) When using the find_bads_ecg() function in ica.py with the arguments "start" and "stop" set to some <tmin>, <tmax> values and the "method" argument set to <method='CTPS'> the "start", "stop" arguments are not used at all, which in turn results in (too many) ecg events from the entire raw file

2.) in ctps_.py the estimated p value in function _prob_kuiper() get's easily close to zero (or 1 after normalisation) if high synchronization was detected AND a large number of trials (epochs) are used. To prevent for an under- or overflow (i.e., unrealistic perfect synchronization measurement with p=0.0 or pk=1.0) the np.exp() function in line 154 should use "float128". This is tolerable as the array is small. For this, the line should read np.exp(-2. * fact * j2 * l2, dtype='float128'). 

If you want us to do the changes, we are happy to do so.

dengemann, 2017-11-21:
> "start" and "stop" set to some , values and the "method" argument set to

Hi Jürgen!

Thanks for reporting. Your contribution is most welcomed and we’re happy to
help getting the changes in. It also sounds as if it wasn’t so difficult.

Denis

agramfort, 2017-11-22:
yes please send us a PR.

thanks

jdammers, 2017-11-22:
OK, PR is coming very soon.

closed 2021-02-14 by hoechenberger