mne-tools/mne-python #4769: Create annotation within other annotation

cbrnr, 2017-11-22:
Maybe I'm missing something, but since it is possible to create overlapping annotations (with different types) I thought that it should also be possible to create an annotation within some other annotation (of a different type). Apparently, this does not work, because if a start my new annotation in a region where there is already some other annotation, no selection rectangle is created.

Moreover, is it somehow possible to use different colors for different annotations (I didn't find any argument e.g. to `raw.plot`)?

agramfort, 2017-11-22:
I don't get the question. See:

[image]

if it starts with BAD then it stays red.

cbrnr, 2017-11-22:
You can't create a new annotation which is completely within another annotation. This is useful if you don't only consider good/bad, but arbitrary information such as e.g. "trial", "cue", and "feedback". In this example, the "trial" annotation would extend from e.g. 2-10s. If I then want to create a "cue" annotation from e.g. 3-5s, I can't create it.

agramfort, 2017-11-23:
indeed. There should be a way to toggle on/off certain type of annotations

drammock, 2020-11-17:
Not sure when it got fixed, but this works now.

closed 2020-11-17 by drammock