mne-tools/mne-python #4770: Option to scroll in smaller increments

cbrnr, 2017-11-22:
It would be nice to be able to scroll `raw.plot` in smaller increments than a whole page. Currently, this doesn't seem to be possible. The right and left arrow keys scroll right/left by a whole page. Is it possible to map e.g. Shift+right/Shift+left to scrolling right/left by e.g. only 1/10 of a page?

mmagnuski, 2017-11-22:
+2.5, I wanted that pretty often too. Maybe ctrl + arrow (it would actually be more intuitive if simple arrow keypresses had smaller increments than ctrl + arrow, but that would not be convenient)?

cbrnr, 2017-11-22:
I'd rather not change the current behavior. Ctrl+Arrow is problematic on Macs because this shortcut is associated with switching desktops, so I'd prefer Shift+Arrow. The interesting thing would be if Matplotlib supports shortcuts consisting of more than one keystroke...

agramfort, 2017-11-22:
no objection as long as it works on all platforms and it's documented

larsoner, 2017-11-22:
In my UI experience, shift+arrow being "take larger steps" is common. One way we could make it more uniform would be to have some `time_delta`-like argument that is expressed in units of `duration`, the default thus being `1`. You could use `0.1` for your use case (meaning, move 10% of the window duration each arrow press). shift+arrow could then be mapped to e.g. 5x whatever the standard increment is, more in line with how other software treats shift+arrow.

cbrnr, 2017-11-22:
Yeah, I like this idea. Should I start with a PR?

cbrnr, 2017-11-22:
I'd even go through a deprecation cycle to change the default to e.g. 25% or 50%. This is also more similar to what EEGLAB does (AFAIK EEGLAB takes 1/10 steps with each arrow press and a full step with some key combination).

larsoner, 2017-11-22:
For viz functions and interactions we're usually a bit more lenient on the deprecation cycle. +1 for change to 25% from me

cbrnr, 2017-11-22:
Hmmm, Matplotlib doesn't seem to pick up the shift key as a modifier. Ctrl or command works, but when I press shift+right it only registers as right. Any ideas on how to proceed?

larsoner, 2017-11-22:
From very brief reading/searching it looks like `shift` is treated like a normal key not a modifier, so we can track the state internally, similar to this

https://stackoverflow.com/questions/18141069/matplotlib-how-to-pick-up-shift-click-on-figure

cbrnr, 2017-11-22:
Unfortunately this doesn't work. Shift isn't registered at all. Also, even if it was registered, I probably couldn't combine it with another key (the SO example is combining shift with a mouse event).

larsoner, 2017-11-22:
Works for me, what backend are you on?
```
>>> fig = plt.figure()
>>> def on_key(event):
...     print('you pressed', event.key, event.xdata, event.ydata)
>>> fig.canvas.mpl_connect('key_press_event', on_key)
>>> def off_key(event):
...     print('you released', event.key, event.xdata, event.ydata)
>>> fig.canvas.mpl_connect('key_release_event', off_key)
you pressed shift None None
you pressed A None None
you released A None None
you released shift None None
```

cbrnr, 2017-11-22:
Ah - it doesn't work with `osx`, but `qt5` works! Cool, I think we can live with that, another reason to always switch to the `qt5` backend. PR incoming.

PR #4773 (2017-11-24): [MRG] Add horizontal scrolling in smaller steps
Fixes #4770.

I'd like to add an explanation to the docstring - any idea for a short sentence (currently everything I come up with is too convoluted and no one would understand what I mean)?
closed 2017-11-24 by larsoner