mne-tools/mne-python #4771: Green vertical line at t=0 in raw.plot

cbrnr, 2017-11-22:
I was wondering what the vertical green line at t=0 indicates. This should be documented somewhere.
[image]

mmagnuski, 2017-11-22:
I am not sure whether this line is of much use once we have annotations. It would be probably more useful to have topomap for specified time point pop up when one right-clicks the signal. This is one of the EEGlab functionalities that I still sometimes miss especially when teaching how to identitfy various artifacts in the signal.

cbrnr, 2017-11-22:
So I take it that you don't know what this line means either 😄 ? I always thought it indicates a non-zero `first_samp`, but this is not true because my data starts at t=0 and `first_samp=0`. If it only indicates the start of the file (sample 0), I suggest we remove it because for me it was more confusing than helpful.

agramfort, 2017-11-22:
no idea :)

larsoner, 2017-11-22:
It was just a way to mark a time point. It's useful when you scroll left and right. It used to be (maybe still is?) that if you clicked in the plot somewhere not on a trace, it would move the green line there.

cbrnr, 2017-11-22:
Thanks @larsoner, this makes sense. However, in a multichannel recording with lots of channels on a page it is almost impossible to click somewhere not on a trace, so practically it is impossible to move this line. I think a dedicated shortcut (e.g. Shift+Click) could solve this issue.

larsoner, 2017-11-22:
> I think a dedicated shortcut (e.g. Shift+Click) could solve this issue.

Sure, or make it possible (and document) that you can e.g. click along the x-axis labels to move the line, too.

cbrnr, 2017-11-22:
This is even better. No need to introduce unnecessary shortcuts no one remembers. 👍

larsoner, 2018-10-04:
@cbrnr can you take a look at implementing this in the next two weeks to make it into 0.17?

mmagnuski, 2018-10-04:
I thought it worked by right clicking, so there was not problem with clicking on channels but I may be wrong. Anyway it would be cool to be able to spawn topoplots for a time point, not only the green line via right-click.

cbrnr, 2018-10-05:
Right-clicking seems like a good option. Currently, it doesn't work, maybe it used to and now it is broken?

cbrnr, 2018-10-08:
I've never used the green line either. I don't have an opinion on showing topos, if this is something people would use then we should implement it. However, we need to discuss what will happen on a right click - moving both the green line and showing a topo is not a good option.

mmagnuski, 2018-10-08:
@cbrnr - I don't think it is not a good option if no one actually uses the green line itself. Showing topographies and the time point that they represent makes the green line useful (the line could disappear ones topos are closed for example). But I agree that it would be safer to allow to mark a time point with rightclick and separately (after pressing `t`, as in brainstorm, for example) open topographies window for that marked timepoint.

massich, 2018-10-16:
I don't, for me it works in master. 
```py
from mne.viz.tests.test_raw import _get_raw
[ _get_raw().plot() for _ in range(2)]
```
left click somewhere in one of them, and the green line moves. See:
[image]

PR #5582 (2018-10-16): [MRG+1] Bring back the green line
Fixes #4771.

This (re)activates the vertical green line, which can be used to mark a specific position in time.

Current problems:
1. When right-clicking on a trace, both the green line is moved and the channel corresponding to the trace is marked as bad. Only the green line should be moved.
2. The green text indicating the current position of the line is too close to the x axis labels - is there any way to move the text more to the left?

@larsoner I think you are familiar with the plotting code, could you take a look? I'm not sure if what I did is OK or jus [... cut]
commit 0db3c577 (2018-10-16): [MRG+1] Bring back the green line (#5582)

Fixes #4771.

This (re)activates the vertical green line, which can be used to mark a specific position in time.

Current problems:
1. When right-clicking on a trace, both the green line is moved and the channel corresponding to the trace is marked as bad. Only the green line should be moved.
2. The green text indicating the current position of the line is too close to the x axis labels - is there any way to move the text more to the left?
closed 2018-10-16 by massich