mne-tools/mne-python #4774: passing two eye channels as arguments

aitorme, 2017-11-23:
I ran an EEG experiment with 32 channels, and when I want to run `mne.preprocessing.find_eog_events`, I need to specify the eye channels I used since the default is for an EEG with 64 channels. My problem is that I cannot pass two arguments for the `ch_name` option of the function, and trying to pass both at the same time as a list or tuple does not work either.

agramfort, 2017-11-23:
you should pass the vertical EOG channel or make a virtual EOG from a
derivation.

aitorme, 2017-11-23:
is there any documentation to make the derivation of that? I'd rather have the information from both eye channels if possible

teonbrooks, 2017-11-24:
if you have generated a `Raw` object, you can use the method `set_channel_types` to assign the specified channels as eog. also, at the file reading, you can specify eog channels with the parameter 'eog'.

aitorme, 2017-11-28:
I cannot find any documentation for `set_channel_types`; what would be the difference between using that function and specifying them when importing the `raw` object at file reading?

mmagnuski, 2017-11-28:
@aitorme 
For example: https://martinos.org/mne/stable/generated/mne.io.RawArray.html#mne.io.RawArray.set_channel_types

teonbrooks, 2017-11-29:
> what would be the difference between using that function and specifying them when importing the raw object at file reading?

there's no real difference between setting the eog at the `Raw` initialization and afterwards

aitorme, 2018-02-23:
thanks for your help guys!

closed 2018-02-23 by aitorme