mne-tools/mne-python #4779: "Significant differences" for plot_compare_evoked

jona-sassenhagen, 2017-11-24:
As suggested by @kingjr :

>Don't know if it would be useful or over-the-top, but in my plotting functions, I typically add an optional sig parameters to highlight the significant parts of the curve.
When confronted to multiple curves I bold the curves that are sig. different from chance
fill between the two curve when the two curves are sig. from one another
e.g. example
https://cloud.githubusercontent.com/assets/4881164/16731694/74e20ca8-4747-11e6-8507-f1495430024d.png

I can imagine this being useful. If people agree, I would go about implementing it.

kingjr, 2017-11-24:
> 4747-11e6-8507-f1495430024d.png

I'm not sure. In my opinion, we should teach people on how to make these
kind of plots, and master matplotlib, else we'll run into
specificity/support issues. But that's just a feeling

jona-sassenhagen, 2017-11-24:
You're turning into Alex :|

kingjr, 2017-11-24:
I take this as a compliment ;)

agramfort, 2017-11-25:
+1 :)

fully agreed with JR

mmagnuski, 2017-11-25:
I also agree, although it could be useful to provide some _simple_ function(s) for highlighting portions of 1d / 2d data I wouldn't build it into any existing function - rather show in a tutorial how to write such highligting code.

dengemann, 2017-11-25:
Could not agree more. Teach people to use matplotlib. They have the
bandwidth to actually support viz and its documentation.

jona-sassenhagen, 2017-12-09:
Thanks for the input!

closed 2017-12-09 by jona-sassenhagen