mne-tools/mne-python #4780: KIT layout for plots

VedaHung, 2017-11-27:
Hi MNE experts,

I used KIT system for MEG data recording.
MEG data were imported to mne with two following functions:
raw = mne.io.kit.read_raw_kit(file, stim = [200],preload=True)
kit_lay = mne.channels.layout.read_layout('KIT-TW-157.lout')

However,I had problems with two following functions:

(1) ica.plot_properties(epochs, i)
      error message: "the KIT system with ID 261 is missing"

(2) mne.viz.plot_evoked_topo(evokeds, layout=kit_lay)
     no error message but caused the program to crash 
A sample MEG file (MEG160_sample.zip) is attached here. 
Your suggestions will be appreciated. Thanks!

Best,
Veda

[MEG160_sample.zip](https://github.com/mne-tools/mne-python/files/1504660/MEG160_sample.zip)

larsoner, 2017-11-27:
@christianbrodbeck any ideas?

teonbrooks, 2017-11-27:
@VedaHung I was able to reproduce the error. since I don't have the layout to see if this remedies the problem, try the following:

```python
ica.plot_properties(epochs, i, topomap_args=kit_lay)
```

teonbrooks, 2017-11-27:
@VedaHung I was able to reproduce the error. since I don't have the layout to see if this remedies the problem, try the following:

```python
ica.plot_properties(epochs, i, topomap_args={layout: kit_lay})
```

VedaHung, 2017-11-28:
Hi,
Thanks for the suggestions. However, adding KIT layout in  topomap_args did not work. 
BTW, I attached the layout I used here.
Veda

[KIT-TW-157.zip](https://github.com/mne-tools/mne-python/files/1508112/KIT-TW-157.zip)

VedaHung, 2017-11-28:
Hi all,

Someone suggested to use the function as following and it workded.
ica.plot_properties(epochs, i, topomap_args={'layout': kit_lay})

Veda

christianbrodbeck, 2017-11-28:
To make it easier, the layout could be added to the builtin layouts, it just needs to be registered to the appropriate system-ID in `mne.io.kit.constants.KIT_LAYOUT`.

larsoner, 2018-04-04:
Closing as there is a suitable workaround, but if someone figures out which layout this KIT system should use we can open a PR to add it to the list

closed 2018-04-04 by larsoner