mne-tools/mne-python #4784: raw.plot_psd(reject_by_annotation=False) crashes

cbrnr, 2017-11-27:
When I run `raw.plot_psd(reject_by_annotation=False)` on a `raw` object containing annotations, I get the following error:

```
/usr/local/lib/python3.6/site-packages/mne/io/base.py in get_data(self, picks, start, stop, reject_by_annotation, return_times)
    890             data, times = self[picks, start:stop]
    891             return (data, times) if return_times else data
--> 892         if reject_by_annotation.lower() not in ['omit', 'nan']:
    893             raise ValueError("reject_by_annotation must be None, 'omit' or "
    894                              "'NaN'. Got %s." % reject_by_annotation)

AttributeError: 'bool' object has no attribute 'lower'
```

The problem is [here](https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/psd.py#L55):

    rba = 'NaN' if reject_by_annotation else reject_by_annotation

If `reject_by_annotation` is `False`, it remains `False`, but afterwards the string method is called on this `bool` object and fails.

larsoner, 2017-11-27:
This indeed seems like bad checking, PR welcome if you have time

mmagnuski, 2017-11-28:
I've had that too lately, should be an easy fix. BTW - does `reject_by_annotation=None` work for you?

cbrnr, 2017-11-28:
@mmagnuski yes, `None` works, and so does *any* string.

cbrnr, 2017-11-28:
It seems we only want to accept `True` or `False`. The question is, where should these arguments be checked? We could do this in multiple places.

larsoner, 2017-11-28:
If we only allow bool in `plot_psd`, do it there. If we do it multiple places, add a helper to `mne/viz/utils.py` or `mne/utils.py` (as appopriate based on where the functions that will use it live) and use it in those places

cbrnr, 2017-11-28:
The call stack is `mne.io.BaseRaw.plot_psd` -> `mne.viz.raw.plot_raw_psd` -> `mne.time_frequency.psd.psd_welch` -> `mne.time_frequency.psd._check_psd_data`. The error occurs in the last function, because here `True` or `False` needs to be converted to `None`, `'omit'` or `'NaN'`. So I guess this line

    rba = 'NaN' if reject_by_annotation else reject_by_annotation

should be changed to

    rba = 'NaN' if reject_by_annotation else None

Correct?

larsoner, 2017-11-28:
Sounds reasonable to me

PR #4790 (2017-11-29): [MRG] Fix raw.plot_psd reject_by_annotation=False
Fix #4784
closed 2017-11-29 by larsoner