mne-tools/mne-python #4788: Is scikit-learn required for ICA?

cbrnr, 2017-11-28:
When creating an `ICA` object, I get the following error if scikit-learn is not installed:

    RuntimeError: the scikit-learn package (version >= 0.12)is required for ICA

I think MNE's ICA implementation uses the `FastICA` class from scikit-learn, but AFAIK it ships its own (Extended) Infomax algorithm, in which case scikit-learn should not be required.

cbrnr, 2017-11-28:
Also, there's a space missing in this error message. And the main README says that scikit-learn >= 0.18 is required - which version is it?

larsoner, 2017-11-28:
>  AFAIK it ships its own (Extended) Infomax algorithm, in which case scikit-learn should not be required.

We also use randomized PCA from `sklearn` for dimensionality reduction and whitening.

> the main README says that scikit-learn >= 0.18 is required - which version is it?

I think the error message is old and should be updated to say 0.18

PR #4789 (2017-11-29): [MRG+1] Require sklearn >= 0.18
Fix #4788
closed 2017-11-29 by larsoner