mne-tools/mne-python #4792: Selecting ICA components in `ica.plot_components` doesn't work anymore

cbrnr, 2017-11-29:
If I click on the title of a component in the dialog produced by `ica.plot_components(inst=raw)`, I get the following error:

```
  File "/usr/local/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 389, in process
    proxy(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 227, in __call__
    return mtd(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/topomap.py", line 1006, in onclick
    label = event.inaxes.get_label()
AttributeError: 'NoneType' object has no attribute 'get_label'
```

Any idea why the label is gone and how to retrieve it (actually, `event.inaxes` is `None`)?

mmagnuski, 2017-11-29:
I'll take a look in the evening, which matplotlib version do you have?

cbrnr, 2017-11-29:
Thanks! I have the latest Matplotlib (2.1.0).

mmagnuski, 2017-11-29:
What is your expectation for what should happen when you click the title? I can replicate your problem but if you want to open component properties then click on the topo, not the title. Then this would be an easy fix.

cbrnr, 2017-11-30:
Clicking on the title should mark/unmark a component as bad (i.e. include/exclude in `ica.exclude`).

mmagnuski, 2017-11-30:
Ah, right, that's something that you've added IIRC. Where is that set up in the code (I looked but couldn't find it)? Because the error comes from the piece of code that plots properties, not tries to change `ica.exclude`.

cbrnr, 2017-11-30:
What, I added this? :open_mouth:

mmagnuski, 2017-11-30:
Correct me if I'm wrong. :) I looked into PR history and found this related PR: #4285, but it does only change title coloring, no interactivity is added. When was the title-related interactivity added?

cbrnr, 2017-11-30:
AFAIR I wanted to add interactivity but then we decided to quickly merge this PR with only the color change implemented. So even if I did not yet add interactivity we still need to handle this error, right? We should be able to click on the title and nothing should happen.

mmagnuski, 2017-11-30:
Yes, that was why I mentioned that it should be an easy fix if clicking on titles doesn't have any intended functionality now. I will push a PR today. I can also add the selection by title clicking, unless you want to do it. :)

cbrnr, 2017-11-30:
Actually, if you want to add the interactive functionality I'd be more than happy! :smile:

mmagnuski, 2017-11-30:
Ok, I'll try to do it in one PR.

mmagnuski, 2017-11-30:
Because clicking on the title will always give you inaxes of None, the selection (changing state from excluded to included and the reverse) can be done in an easy or hard way:
* *easy* - right-clicking the topo changes IC state
* *harder* - clicking the title changes ICA state, but that would require checking the bounding boxes of titles with respect to xy pos of the mouse click.

Because of this I will split this into two PRs. First - small fix so that inaxes of None is ignored and only when we decide which of the above scenario we want I'll add it in another PR.

mmagnuski, 2017-11-30:
BTW anyone knows how to quickly find all the title text objects? In this case I can do it by inspecting all text object and seeing which have a text beginning with `'ICA'` but I was wondering if there is an easier way.
Once this is done we could check all text objects with their `contains` method against the mouse event.

larsoner, 2017-11-30:
Instead of inspecting all, we know there is only one we care about. So it might be easier to know which `Axes` instance you're interested in and check only the `contains` of its title text object as `ax.title` IIRC.

cbrnr, 2017-11-30:
> Because clicking on the title will always give you inaxes of None

Did this change recently? AFAIR, you could click on the title and nothing happened.

From a UX perspective, the harder option (clicking on the title) would be preferable. In plot_raw, you can click on the y-axis channel labels to select/deselect a channel - is this something similar you could use?

larsoner, 2017-11-30:
(and if you don't know which, iterate over `fig.axes` and find the one with an appropriate `ax.get_title()` str)

mmagnuski, 2017-11-30:
@larsoner
Cool, I forgot about `ax.title` :) But this would sitll require testing all titles: if the user clicks on title I only know the x and y of the mouse.

larsoner, 2017-11-30:
I'm thinking:
```
for ax in fig.axes:
    if ax.get_title().startswith('ICA '):
        # check ax.title
```

mmagnuski, 2017-11-30:
Yes, this is similar to my line of thinking :) But we can gather the titles beforehand - during the construction of ic topos. So I actually meant "all ic titles".

larsoner, 2017-11-30:
Yeah that should be fast enough

mmagnuski, 2017-11-30:
@cbrnr 
> Did this change recently? AFAIR, you could click on the title and nothing happened.

I'm not sure, I never clicked the titles before. But I tried this on a previous matplotlib version and the error was printed in the console where I started jupyter notebook, but was absent in the notebook view. Maybe matplotlib changed something so now all the errors are passed to the notebook (that's actually helpful).

PR #4798 (2018-02-05): [MRG] ENH+FIX: select excluded components by clicking on the ic title
@cbrnr 
This is already contains title interactivity (was easy once @larsoner suggested using `ax.title`). Check if that works for you. We can consider whether gray is visible enough, but I think it is ok.
[WIP] because I'll add a test once you approve how it works in general.

edit: closes #4792
closed 2018-02-05 by larsoner