mne-tools/mne-python #4813: Filter (mute/disable) specific warnings

cbrnr, 2017-12-05:
How can I filter a specific warning I don't want to see? The problem is that warnings go through `warnings` and the `logging` module. Specifically, I want to mute the warning that my ICA files do not follow MNE naming conventions. I can mute the warning (in fact, I want to mute all warnings related to MNE naming conventions) like this:

    warnings.filterwarnings("ignore", ".*not conform to MNE naming conventions.*")

But the logging warning is still there. What is the recommended way to turn off specific warnings in MNE?

larsoner, 2017-12-05:
Right now you can globally use `verbose='error'` or `with mne.utils.use_log_level('error'):` context handler, but it will squash all warnings and infos.

To do what you want I guess we could extend `use_log_level` with an `ignore` regex. Then we'd need to modify (create?) our own `StreamHandler` to filter via regular expression matching.

cbrnr, 2017-12-05:
Yeah, I don't want to globally disable all warnings and infos. Your idea sounds great, do you think this would be useful to have (or am I the only one with this use case)? I'm not sure I have the time and background to implement this quickly.

larsoner, 2017-12-05:
I haven't needed to split the messages, but maybe others do.

I think basically you'd set up a context manager (`__enter__` and `__exit__`) that stored the old handlers and set a new subclassed [StreamHandler](https://docs.python.org/3/library/logging.handlers.html#logging.StreamHandler) that `emit`s only if the regex is satisfied. See this for inspiration about how to access the `mne` logger and its handlers:

https://github.com/mne-tools/mne-python/blob/master/mne/utils.py#L1265

One thing you'd want to note is that it will temporarily disable logging to a file, unless you also want to implement that (but I'd assume YAGNI).

cbrnr, 2017-12-05:
Hm, here's what I came up with:

```
import logging
import re

logger = logging.getLogger("mne")
logger.addFilter(lambda s: not re.match(".*convention.*", s.getMessage()))
```

It seems to work, but now the question is where do we add this functionality? Do you want to add a new `ignore=None` parameter for `set_log_level`?

larsoner, 2017-12-05:
I'd add an `ignore` parameter to the `use_log_level` context handler

cbrnr, 2017-12-05:
And how do I ignore a specific message in my script then? I don't want to wrap everything in a `with` block.

larsoner, 2017-12-05:
You would only need to wrap the offending call (usually `read_raw_fif`). Is this too much?

cbrnr, 2017-12-05:
Hm, I also have a `read_ica` call which triggers this warning. Ideally, I'd set it at the top of my script with `mne.set_log_level('warning', ignore='.*convention.*')`, but I guess I could also wrap the offending calls directly. What about the warnings emited by `warn` though? Do I still need to filter them separately?

cbrnr, 2017-12-05:
Or alternatively, can't we downgrade the message related to file naming conventions to info (and not emit a `warn`)? This would also solve my problem :smile:.

larsoner, 2017-12-05:
I think the `warn` function that we use checks the `logging` level and if it's set such that `logger.warn(...)` is ignored it does not emit a `warnings.warn(...)` either, but it would be good to double-check

cbrnr, 2017-12-05:
You're right, `warn` does take care of that. Still, are you open to downgrading this to info?

larsoner, 2017-12-05:
Soon we're going to have to relax or change our naming checks to BIDS formatting guidelines anyway.

At least for raw FIF data I don't mind relaxing the checks. But would you also want them changed or removed for ICA (and evoked, cov, etc.)? The tradeoff is between trying to improve standard adoption, and not annoying users. So if we can offer users a way of not being annoyed (`verbose='error'` or more fine control via `use_log_level`) it seems preferable. I suspect a lot of people use `warning` log level, so `info` message as suggestions toward standarization will be missed.

cbrnr, 2017-12-05:
The question is how are standard file name conventions enforced? I prefer to use `*_ica.fif` instead of `*-ica.fif` because all my files are separated by underscores. Does anything stop working if I don't follow the convention? If not, this message should be an info and not a warning. And yes, I use a `warning` log level, but these warnings spam my output which might make me miss important warnings.

larsoner, 2017-12-05:
> Does anything stop working if I don't follow the convention? If not, this message should be an info and not a warning.

I'm not convinced. There is value in us trying to get the community to follow standardized naming conventions. I think we should offer people a way to avoid these warnings if they don't want them. So far `verbose='error'` has probably been enough for people, and the API we've talked about here would give finer grained control.

For the specific case of ICA I'm fine with adding `_ica.fif` to the list of accepted names.

cbrnr, 2017-12-05:
Yes, enforcing standards is a good thing, I agree. I'm just not sure if a context handler is the best option. Personally, I like how `warnings.filterwarnings` lets you globally filter out specific messages. I would prefer a similar function in MNE (`mne.filterwarnings`?) that filters both warnings and logging messages at the warning level. This way, users could disable specific messages with one line of code.

cbrnr, 2017-12-05:
OK, now I get it. You mean I should just use `read_ica('whatever_file', verbose='error')`? Of course this is also fine!

larsoner, 2017-12-05:
> You mean I should just use read_ica('whatever_file', verbose='error')? Of course this is also fine!

Yes that is what I do. But in cases where you care about the `logger.info` (like you might in `read_raw_fif`) then you would need some finer level of control like we've been discussing.

> I'm just not sure if a context handler is the best option. Personally, I like how warnings.filterwarnings lets you globally filter out specific messages. I would prefer a similar function in MNE [...]

I slightly prefer the context manager because it limits the scope of changes. This is less convenient than the `filterwarnings` approach, but at the same time more explicit about where warnings will actually be ignored, which otherwise needs to be inferred (and not necessarily correctly based on the regex complexity!) in the global case.

However, adding the context manager is probably a dozen lines of code once a global-level solution is implemented (assuming you implement an `ignore=None`-means-"reset" option), so feel free to implement that instead -- we can always add the context manager later.

cbrnr, 2017-12-05:
> Yes that is what I do. But in cases where you care about the `logger.info` (like you might in `read_raw_fif`) then you would need some finer level of control like we've been discussing.

Unfortunately, `read_ica` doesn't accept a `verbose` argument, so it would be the context manager again.

> I slightly prefer the context manager because it limits the scope of changes. This is less convenient than the filterwarnings approach, but at the same time more explicit about where warnings will [...]

Both approaches have pros and cons and it often depends on the situation which one is better. I don't have a solution here, only I still prefer the global solution in my use case.

> However, adding the context manager is probably a dozen lines of code once a global-level solution is implemented (assuming you implement an ignore=None-means-"reset" option), so feel free to implement that instead -- we can always add the context manager later.

OK, I'll try that.

But before I do, can we discuss why non-conventional names trigger warnings again? I thought about it some more, and a warning should indicate something important that directly influences something, e.g. if ICA hasn't converged or something a warning would be appropriate because the users needs to be aware that the solution might be sub-optimal. In the case of non-conventional file names, what are the ramifications? To me, these should be infos - still important to know (in a REPL), but not worth emitting in a script over 50 people.

larsoner, 2017-12-05:
> Unfortunately, read_ica doesn't accept a verbose argument, so it would be the context manager again.

[FTFY](https://github.com/mne-tools/mne-python/commit/dc9d4da05cd18dcd0ce43ddfc6b8cb40bc8dea8d) -- this was just an oversight as it already even had the `verbose` decorator

> a warning should indicate something important that directly influences something

Even under this definition my arguments hold if you consider standardization as important and directly influences interpretability of what data a given set of files contain upon looking at them as files in a filesystem.

The report module currently relies on these names when doing `parse_folder`, although in theory we could write a heuristic "what type of FIF file is this" function that could probably figure out what type of file each `.fif` file was.

FWIW it looks like the checks have been in the software since [0.8](https://github.com/mne-tools/mne-python/commit/49718885f1f912a46627a4f308ad75f9f3747581) (3.5 years and 7 releases). In that time I only recall one other user complaining about it (@kingjr at one point?) so I suspect the "visible gentle pushes" toward standardization have been worth the inconvenience of `verbose='error'` or similiar workarounds.

I'll let others comment, though. If there is consensus that it's not worth the tradeoffs we can scrap it.

cbrnr, 2017-12-06:
> FTFY -- this was just an oversight as it already even had the verbose decorator

Thanks! Isn't the `verbose` decorator supposed to add the `verbose` argument? Why is it necessary to explicitly add it to the function?

> Even under this definition my arguments hold if you consider standardization as important and directly influences interpretability of what data a given set of files contain upon looking at them as files in a filesystem.

Again, I don't consider this as important as other warnings related more directly to the outcome of a function. That's why they should be in the info category.

Also, why does `read_ica` trigger this warning in the first place? I agree that it could remain a warning in `ica.save`, but `read_ica` is not responsible for creating correct file names.

FWIW, I will just rename my ICA files to suppress these warnings for now. There you have it. Now I follow the standards 😸 .

I'd be interested in other opinions too. Here are possible options we could go forward with:

1. Everything is fine, stop yelling about that and just rename your files.
2. Messages related to violating file naming conventions should be in the info level and not warnings.
3. Let's just include `..._ica.fif` in the list of acceptable file names.
4. Let's downgrade the warnings to infos in `read_ica`, but let's leave them as warnings in `ica.save`.
5. If these warnings bother you, use `read_ica(..., verbose='error')`.
6. Let's have a way to disable specific warnings in the `use_log_level` context manager.

I'm fine with the first five options based on the outcome of this discussion. Option 6 is too cumbersome, and at this stage of the discussion I don't think it is necessary any more.

@kingjr @agramfort @dengemann @jona-sassenhagen (anyone else?)

kingjr, 2017-12-06:
2

jona-sassenhagen, 2017-12-06:
I am in favour of warnings for violations of standards, but I also think we produce too many errors - it rewards turning them off entirely. Not sure what solution that entails though.

agramfort, 2017-12-06:
I would go with 3 but if you really insist 2 could be fine

cbrnr, 2017-12-18:
OK, it seems we're down to two options:

2. Messages related to violating file naming conventions should be `info` instead of `warning` (@kingjr, @agramfort, myself, (@jona-sassenhagen)).
3. Add `..._ica.fif(.gz)` to the list of acceptable file names (@larsoner, (@jona-sassenhagen)).

I'm happy to implement either one, but someone has to make the decision 🤔 .

larsoner, 2017-12-18:
@agramfort voted for switching to `info` (2)?

cbrnr, 2017-12-18:
He said if I insist... 😛

agramfort, 2017-12-18:
I am reading the thread again and the argument "I prefer to use *_ica.fif instead of *-ica.fif" does not really fly for me. We all have preferences and the strength of stanbdards is that we don't have to teach each other what we all prefer :) remember pep8 right?

@cbrnr can we convince you to use -ica.fif?

otherwise it would mean changing the convention/standard but it should be done everywhere (_epo.fif, _ave.fif etc.)

cbrnr, 2017-12-18:
Oh yeah, I'm fine with adding these to the standard names. I'll create a PR shortly.

agramfort, 2017-12-18:
so you vote for changing the "standard" ?

how about maybe your habits the habits of all of us?

cbrnr, 2017-12-18:
Sorry, read your last message too quickly. Sure, I'll change my files to -ica, although I still think we shouldn't be that pedantic and allow _ instead of -.

larsoner, 2018-04-04:
@cbrnr do you still want this?

larsoner, 2018-04-04:
Closing for now but if you think there is more to discuss feel free to reopen

closed 2018-04-04 by larsoner

cbrnr, 2018-04-11:
I guess it's OK the way it is.

JJKK1313, 2022-05-14:
It wasn't fixed yet. those warnings are very annoying and the `set_log_level` doesn't work. also trying to stream those warnings to a file doesnt work because `stdout` is still being used!