mne-tools/mne-python #4819: Q? morph volume source estimate

kingjr, 2017-12-07:
I normally use `compute_morph_matrix` to  generate a morph that I can apply to my  source estimate `stc.morph_precomputed`.

However, I am now trying to use LCMV beamformers, which generate volume source estimate. Such object do no have `morph_precomputed` methods. Is there a way to morph such volume stc to `fsaverage`?

Thanks!

cc @avkasdan @britta-wstnr

larsoner, 2017-12-07:
FWIW you can do LCMV using a surface-based source space if you want.

For transforming MRI data we don't provide anything but e.g. `nibabel` might. Basically you can use `stc.as_volume()` to get a standard MRI NIfTI file/`nibabel` instance:

https://martinos.org/mne/stable/generated/mne.VolSourceEstimate.html#mne.VolSourceEstimate.as_volume

At that point you have a standard MRI problem of transforming data into a standard subject space (e.g., MNI or fsaverage). Freesurfer (and maybe even e.g. `nibabel`) must provide ways of doing this, but I haven't had to do it yet so hopefully someone else can point you in the right direction.

kingjr, 2017-12-09:
ok , I'll close it, as I doubt anyone will have time to do such ENH.

closed 2017-12-09 by kingjr

larsoner, 2017-12-09:
It would be great if you can report back what solution you come up with in
the end. You might want to email the list as people there might have ideas.

kingjr, 2017-12-09:
I went back to surface stc