mne-tools/mne-python #4821: BUG: metadata update field

kingjr, 2017-12-08:
The user cannot update the metadata field:

```python

import numpy as np
import pandas as pd
epo = mne.Epochs(mne.io.RawArray(np.random.randn(10, 1000),
                                 mne.create_info(10, 1000.)),
                 events=np.array([[100, 0, 1], [200, 0, 1], [300, 0, 1]]),
                 tmin=-.050, tmax=.100,
                 metadata=pd.DataFrame([dict(original_key=idx) for idx in range(3)]))
epo.metadata['new_key'] = 1
print(epo['original_key'])  # works
print(epo.metadata['new_key'])  # does not work
print(epo['new_key'])  # doesnt work
```

cc @choldgraf @jona-sassenhagen

jona-sassenhagen, 2017-12-09:
Yeah, for now, you need to do:

```
df = epochs.metadata
df["x"] = "x"
epochs.metadata = df
```

kingjr, 2017-12-09:
or you can do 
```python
epochs._metadata['new_key'] = 1
```

but we shoulnd't need to go through a hidden var or df replacement IMO

agramfort, 2017-12-09:
agreed going through df aux var seems an overkill for me.

choldgraf, 2017-12-09:
I think the reason for this was to do consistency / error checks any time
something is edited

larsoner, 2017-12-09:
Yes it's to prevent users from e.g. removing rows manually and breaking
everything. I guess we could instead add a private consistency checker
function that we run that checks length (and row index match with
.selection eventually) anytime we want to access metadata

agramfort, 2017-12-09:
sounds like a good idea

PR #4827 (2017-12-12): MRG: Fix metadata
Closes #4820.
Closes #4821.
Closes #4823.

@kingjr can you confirm everything works as you expect now?
closed 2017-12-12 by jona-sassenhagen