mne-tools/mne-python #4822: BUG: default epochs events does not work with list

kingjr, 2017-12-08:
works:

```python
import numpy as np
epo = mne.Epochs(mne.io.RawArray(np.random.randn(10, 1000),
                                 mne.create_info(10, 1000.)),
                 events=np.array([[100, 0, 1], [200, 0, 1], [300, 0, 1]]))
```

does not work:
```python
import numpy as np
epo = mne.Epochs(mne.io.RawArray(np.random.randn(10, 1000),
                                 mne.create_info(10, 1000.)),
                 events=[[100, 0, 1], [200, 0, 1], [300, 0, 1]])
```

cc @larsoner

larsoner, 2017-12-08:
Yes we should cast to `np.array(...)`, check the dtype and shape (raise error if incorrect), and cast to `int`. PR welcome

jona-sassenhagen, 2017-12-09:
We actually convert to an array, but it happens later:

https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L263

We could change this line:
`            event_id = list(np.unique(events[:, 2]))`
so that it works for arrays and lists, e.g. something like
`event_id = list({id for _, _, id in events})`

I don't think `events` will be large enough for this to become a problem.

larsoner, 2017-12-09:
We should just convert and check earlier, your suggested code will yield
less clear errors depending on input

agramfort, 2017-12-09:
see https://github.com/mne-tools/mne-python/pull/4825

PR #4825 (2017-12-10): FIX : Fix bug in mne.Epochs when passing events as list with event_id=None
closes #4822
closed 2017-12-10 by agramfort