mne-tools/mne-python #4823: BUG: metadata index behavior does not comply with pandas

kingjr, 2017-12-08:
the raw index is reset by default. This makes the user lose track of the original indices of the metadata. #bad

```python
import mne
import numpy as np
import pandas as pd
epo = mne.Epochs(mne.io.RawArray(np.random.randn(10, 1000),
                                 mne.create_info(10, 1000.)),
                 events=np.array([[100, 0, 1], [200, 0, 1], [300, 0, 1]]),
                 tmin=-.050, tmax=.100,
                 metadata=pd.DataFrame([dict(original_key=idx) for idx in range(3)]))

epo[1:].metadata  # the raw index is reset by default. 
```

larsoner, 2017-12-08:
[#wetalkedaboutthis](https://github.com/mne-tools/mne-python/pull/4414#issuecomment-331449259)

[#rtfm](https://mne-tools.github.io/dev/auto_examples/preprocessing/plot_metadata_query.html) (though maybe we should make it more prominent)

#amidoingthisright

... basically we lose it during I/O and it would take some thinking (i.e., building something custom on top of the I/O that Pandas natively provides) to get it to be saved. Can you use `epochs.selection` or something else? Can we just advise people to add a column of `np.arange` if they really want the original indices, or is this too much of a hack?

kingjr, 2017-12-08:
> #wetalkedaboutthis

Maybe, my memory isn't great.

In any case, that remain a very bad design. This can cause major analytical errors; might as well have a separate Dataframe.

How about we assign a `_index` column on the fly to store and regenerate the right indices? Then would just have to indicate to the subjects that this column name shouldn't be touched.

```python
pd.DataFrame([dict(trial=ii) for ii in range(10)], index=range(1, 11))
```

larsoner, 2017-12-08:
Can you check to see if it's redundant with `epochs.selection`? If so we can just use that perhaps

kingjr, 2017-12-08:
> Can you check to see if it's redundant with epochs.selection? If so we can just use that perhaps

yes, that seems to be doing the job. So we'd just need to do `pd.DataFrame(metadata, index, epochs.selection)` when subselecting epochs (`epochs[idx:]`), reading and writing.

kingjr, 2017-12-09:
We still need to create the fix. Specifically, anytime we internally read/write/modify the metadata, we need to pass the `epochs.selection` to the metadata index. 

(I don't have the bandwidth right now, sorry)

PR #4827 (2017-12-12): MRG: Fix metadata
Closes #4820.
Closes #4821.
Closes #4823.

@kingjr can you confirm everything works as you expect now?
closed 2017-12-12 by jona-sassenhagen