mne-tools/mne-python #4824: OverflowError: date value out of range

nerdmiao, 2017-12-09:
```
  File "/home/zm/PycharmProjects/braindecode-master/braindecode/datasets/bcic_iv_2a.py", line 13, in load
    cnt = self.extract_data()
  File "/home/zm/PycharmProjects/braindecode-master/braindecode/datasets/bcic_iv_2a.py", line 20, in extract_data
    raw_edf = mne.io.read_raw_edf(self.filename, stim_channel='auto')
  File "/home/zm/mne-python/mne/io/edf/edf.py", line 1232, in read_raw_edf
    exclude=exclude, preload=preload, verbose=verbose)
  File "<string>", line 2, in __init__
  File "/home/zm/mne-python/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/home/zm/mne-python/mne/io/edf/edf.py", line 158, in __init__
    annotmap, eog, misc, exclude, preload)
  File "/home/zm/mne-python/mne/io/edf/edf.py", line 408, in _get_info
    edf_info = _read_gdf_header(fname, stim_channel, exclude)
  File "/home/zm/mne-python/mne/io/edf/edf.py", line 889, in _read_gdf_header
    datetime.timedelta(date * pow(2, -32) - 367)
OverflowError: date value out of range
```
I got this error, how can fix it?

agramfort, 2017-12-09:
you can put a break point eg with:

import ipdb; ipdb.set_trace()

then inspect the value of :

date * pow(2, -32) - 367

to understand what happened.

please show us what:

mne.sys_info()

gives you

nerdmiao, 2017-12-10:
I guessed my dataset files is wrong. I edited it with notepad but it was not aligned。

agramfort, 2017-12-10:
we could make a try except to avoid error and give warnings but we need to
be able to replicate.

can you share a file?

nerdmiao, 2017-12-11:
ok！ It's the data file with incorrect modify.
[A01T.gdf.zip](https://github.com/mne-tools/mne-python/files/1546705/A01T.gdf.zip)

agramfort, 2017-12-12:
thanks

ok I can replicate.

now i tried to see what is expected answer for these lines:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L888

but I see that these lines are not covered by our EDF/GDF tests and I
cannot find
a file on my disk that needs these lines.

@teonbrooks maybe?

what does a matlab reader gives for the date on this file you shared?

nerdmiao, 2017-12-13:
It can not be opened with matlab.

agramfort, 2017-12-13:
it crashes too?

if so I guess it's really a bug with this file so we can close the issue.

nerdmiao, 2017-12-14:
yes.

closed 2017-12-14 by agramfort