mne-tools/mne-python #4828: Reading eeg lab files: Issues with more recent file version

stes, 2017-12-11:
As of matlab 7.3, the `scipy.io.loadmat` interface is not suitable for reading `*.mat` (and thus, `*.set` files) anymore. Attempting to use `mne.read_epochs_eeglab` or `mne.read_raw_eeglab` results in the following error:

```
/usr/lib/python3.6/site-packages/mne/io/eeglab/eeglab.py in read_epochs_eeglab(input_fname, events, event_id, montage, eog, verbose, uint16_codec)
    255     epochs = EpochsEEGLAB(input_fname=input_fname, events=events, eog=eog,
    256                           event_id=event_id, montage=montage, verbose=verbose,
--> 257                           uint16_codec=uint16_codec)
    258     return epochs
    259 

/usr/lib/python3.6/site-packages/mne/io/eeglab/eeglab.py in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, montage, eog, verbose, uint16_codec)

/usr/lib/python3.6/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

/usr/lib/python3.6/site-packages/mne/io/eeglab/eeglab.py in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, montage, eog, verbose, uint16_codec)
    493                  uint16_codec=None):  # noqa: D102
    494         from scipy import io
--> 495         _check_mat_struct(input_fname)
    496         eeg = io.loadmat(input_fname, struct_as_record=False,
    497                          squeeze_me=True, uint16_codec=uint16_codec)['EEG']

/usr/lib/python3.6/site-packages/mne/io/eeglab/eeglab.py in _check_mat_struct(fname)
     41     from scipy import io
     42     mat = io.whosmat(fname, struct_as_record=False,
---> 43                      squeeze_me=True)
     44     if 'ALLEEG' in mat[0]:
     45         raise NotImplementedError(

/usr/lib/python3.6/site-packages/scipy/io/matlab/mio.py in whosmat(file_name, appendmat, **kwargs)
    254 
    255     """
--> 256     ML, file_opened = mat_reader_factory(file_name, **kwargs)
    257     variables = ML.list_variables()
    258     if file_opened:

/usr/lib/python3.6/site-packages/scipy/io/matlab/mio.py in mat_reader_factory(file_name, appendmat, **kwargs)
     69         return MatFile5Reader(byte_stream, **kwargs), file_opened
     70     elif mjv == 2:
---> 71         raise NotImplementedError('Please use HDF reader for matlab v7.3 files')
     72     else:
     73         raise TypeError('Did not recognize version %s' % mjv)

NotImplementedError: Please use HDF reader for matlab v7.3 files
```

Which is thrown by `scipy.io.loadmat`.

Is there an easy way to solve this issue?

agramfort, 2017-12-12:
yes load your files in matlab and save them again to disk with the "-v6"
option.

jona-sassenhagen, 2017-12-16:
Should we maybe catch the error and instead tell people to re-save with "-v6"?

larsoner, 2017-12-16:
In my experience reading modern `.mat` files directly with `h5py` is not so terrible. Has anyone tried it with EEGLAB files? It might not be so difficult to make it work.

We could make a test dataset by loading the existing test `.mat` file in MATLAB, save it out in the new format, read both old and new in MNE and ensure the objects are the same.

stes, 2017-12-16:
There's actually an open PR #4056 concerning this. It would be very convenient if that could be integrated (in particular as the switch to the newer file version will become more important in the future anyway).

cbrnr, 2018-03-30:
I don't think this is super urgent. The default option in EEGLAB is still to save data in the old format. People will have to actively change this setting to save in the new format, which is really only necessary if you want to save data larger than 2GB.
[image]
What I mean is that it would be nice to have, but this should probably not be a blocker for 0.16.

larsoner, 2018-10-04:
This should be fixed in `master`

closed 2018-10-04 by larsoner