mne-tools/mne-python #4829: events not listed in raw object, and unable to add_events, but can find_edf_events

vapniks, 2017-12-12:
I have a raw edf data object, which appears to have no events: 
```
In [1]: data = mne.io.read_raw_edf("data.edf",verbose=True,preload=True)
In [1]: data.info['events']
Out[1]: []
```
even if I try adding some:
```
In [2]: events = np.array([np.arange(0,100000,1000),np.zeros(100),[1]*100]).transpose()
In [2]: data.add_events(events)
In [2]: data.info['events']
Out[2]: []
```
However, the data has some events in the header:
```
In [3]: data.find_edf_events()
Out[3]:
[[35.558594, 2.621094, 'test block'],
 [182.128906, 0, 'Pause -  0:03:03'],
 [220.21875, 1.386719, 'test block 2'],
 [269.691406, 0, 'eyes closed#F1'],
 [284.878906, 0, 'eyes closed#F1'],
 [356.128906, 0, 'Pause -  0:05:57'],
 [365.691406, 0, 'blink22#F2'],
 [368.316406, 0, 'blink22#F2'],
 [371.191406, 0, 'blink22#F2'],
 [374.066406, 0, 'blink22#F2'],
 [376.253906, 0, 'blink22#F2'],
 [381.003906, 0, 'blink22#F2'],
 [384.128906, 0, 'blink22#F2'],
 [387.753906, 0, 'blink22#F2'],
 [391.378906, 0, 'blink22#F2'],
 [396.503906, 0, 'blink22#F2'],
 [400.941406, 0, 'blink22#F2'],
 [405.691406, 0, 'blink22#F2'],
 [415.253906, 0, 'blink22#F2'],
 [418.441406, 0, 'blink22#F2'],
 [421.066406, 0, 'blink22#F2'],
 [424.066406, 0, 'blink22#F2'],
 [427.003906, 0, 'blink22#F2'],
 [461.378906, 0, 'Pause -  0:07:42']]
```
Is this a bug? What's going on?

agramfort, 2017-12-13:
well the confusion come from the fact that events and different from EDF
events
and data.info['events'] should probably not be there...

when you do add_events you create MNE events that get put in the stim
channel
that you can then find with mne.find_events(raw).

find_edf_events allows to get EDF events as present in the original edf
files.

and so data.info['events'] should not be considered.

hope this helps

vapniks, 2017-12-23:
It's a bit awkward that the events returned by find_edf_events cannot be added directly since they are in the wrong format. You have to first translate the times into sample numbers, and the event texts into integers.

agramfort, 2017-12-23:
maybe we could add a helper function?

edf_events are closer to what we call annotations so maybe we could
expose them as annotations and have a way to convert annotations onsets
to events and event_id?

vapniks, 2017-12-25:
As long as they can be used for creating epochs that would be great. Some method for assigning integers to event strings is needed. Maybe by lexicographic order by default, but with the option of specifying the mapping?

agramfort, 2017-12-26:
see annot / annotmap parameter in read_raw_edf

maybe it already does the job for you.

vapniks, 2018-01-06:
OK, thanks

agramfort, 2018-01-06:
does it do the job?

vapniks, 2018-01-06:
Haven't tried it. I wrote a small function to do that, but next time I will have a look at annotations.

agramfort, 2018-01-07:
ok. I would be good if you can share what you have so next time someone hits
this problem is has a solution. Maybe it would be nice to also improve our
documentation.

vapniks, 2018-01-25:
Here is what I did. 
First I wrote a short function to translate event names to numbers (using just the first letters of each name):
```
def get_event_type(name):
    if name[0]=='t':
        return(0)
    elif name[0]=='P':
        return(1)
    elif name[0]=='e':
        return(2)
    elif name[0]=='b':
        return(3)
    else: return(4)
```
then I use that function in a list comprehension to create the new events array:
```
origevents = data.find_edf_events()
events = np.array([[data.time_as_index(t)[0], 0, get_event_type(j)] for (t,i,j) in origevents])
```
then save a copy, just in case
```
mne.write_events('data/events-eve.fif',events)
```
To plot the events, I first create some labels:
```
eventid = {'Test block':0,'Pause':1,'Eyes closed':2,'Blink':3}
fig = mne.viz.plot_events(events, sfreq = data.info['sfreq'], first_samp = data.first_samp, show = False, event_id = eventid)
fig.set_facecolor('white')
fig.axes[0].set_title('Events')
```
The events and eventsid objects can then be used to create epochs:
```
epochs = mne.Epochs(data, events = events, event_id = eventid, tmin = -10, tmax = -5, baseline = None, preload = False)
```

vapniks, 2018-01-25:
Unfortunately I'm bound by an NDA, but I can say it is very simple to create a small function to translate event names to numbers and use that to create a usable events object.

mmagnuski, 2018-01-25:
@agramfort I also got confused by `data.info['events']` when I was beginning with mne python, maybe this field should be removed if it should not be used to store events?

larsoner, 2018-01-25:
We could probably rename it to something else. It's probably worth it to reduce confusion.

mmagnuski, 2018-01-25:
@larsoner What is the field used for? I never found out. :)

larsoner, 2018-01-25:
I don't know either. Probably some legacy Elekta thing.

agramfort, 2018-01-25:
no idea either...

the edf IO should be simplified with Annotations. Cf comments from 2 weeks
ago

larsoner, 2019-08-21:
Closing this as something we won't fix as it could cause backward compat problems, and in our Info docs we now say "..In general this should not be used...".

closed 2019-08-21 by larsoner