mne-tools/mne-python #4834: cross_val_multiscore uses wrong cross-validation methods during running SlidingEstimator or GeneralizingEstimator in classification tasks

acercyc, 2017-12-15:
Based on the source code, **cross_val_multiscore** uses [`is_classifier(estimator)`](https://github.com/mne-tools/mne-python/blob/876192598da7c557dbf46a6dda2ced4b7088d4bb/mne/fixes.py#L1067) to assess the type of the input estimator and then create a cv object. 

[By default](https://github.com/mne-tools/mne-python/blob/master/mne/decoding/base.py#L436), if the input cv value is int (e.g. int(10)), and input estimator is a "classifier", it should create a **StratifiedKFold** cv object. 

### **However,**
when using `SlidingEstimator` or  `GeneralizingEstimator` this kind of "second order" estimators as the input estimator [(e.g. the MNE tutorial here)](http://www.nmr.mgh.harvard.edu/mne/dev/auto_tutorials/plot_sensors_decoding.html#temporal-decoding
), **cross_val_multiscore** can not correctly retrieve the estimator types.

This is because this kind of "second order" estimators do not have the `_estimator_type` attribute. 
This causes **cross_val_multiscore** misinterprets classifiers as regressors and creates an inappropriate cv object. 

By default, it creates a **KFold** (and without shuffling!!) object instead of **StratifiedKFold**.
This inappropriate cv may cause some serious problems in some situations (small sample size, trial has been sorted, ect.).

agramfort, 2017-12-15:
hum indeed.

So SlidingEstimator and GeneralizingEstimator should be added the _
estimator_type attribute
as a property that's it right?

if so can you open a PR?

kingjr, 2017-12-15:
+1 for opening a PR

kingjr, 2017-12-16:
Thanks for noticing, I opened a fix PR

>  By default, it creates a KFold (and without shuffling!!) object instead of StratifiedKFold.
This inappropriate cv may cause some serious problems in some situations (small sample size, trial has been sorted, ect.).

By default, `KFold` or `StratifiedKFold` are not shuffled. We are not going to change this, it's sklearn API. Shuffling can also create problems, so it is up to the user to know what should be done.

acercyc, 2017-12-18:
Thanks for the quick reply and opened a PR.

> By default, KFold or StratifiedKFold are not shuffled. We are not going to change this, it's sklearn API. Shuffling can also create problems, so it is up to the user to know what should be done.

Agree. If it is possible, I suggest this point can be mentioned in the function/class reference and tutorials.
That can help users to notice the default setting and avoid potential problems.

kingjr, 2017-12-18:
Sure, PR welcome to improve the documentation of the tutorials/examples.

PR #4838 (2017-12-18): fix _estimator_type for cross_valmultiscore
closes #4834
closed 2017-12-18 by larsoner