mne-tools/mne-python #4835: RuntimeError: No digitization points found

Stoffander, 2017-12-15:
Dear all,

I ran in a little problem here and could not figure out how to solve it. 

after running:

ica.plot_properties(raw, picks=0)

I get the error message:
``
WARNING:root:Did not find any electrode locations the info, will attempt to use digitization points instead. However, if digitization points do not correspond to the EEG electrodes, this will lead to bad results. Please verify that the sensor locations in the plot are accurate.

RuntimeErrorTraceback (most recent call last)
<ipython-input-86-93ae4d2ec208> in <module>()
----> 1 ica.plot_properties(raw, picks=0)

C:\Users\chris\Anaconda2\lib\site-packages\mne\preprocessing\ica.pyc in plot_properties(self, inst, picks, axes, dB, plot_std, topomap_args, image_args, psd_args, figsize, show)
   1433                                    topomap_args=topomap_args,
   1434                                    image_args=image_args, psd_args=psd_args,
-> 1435                                    figsize=figsize, show=show)
   1436 
   1437     @copy_function_doc_to_method_doc(plot_ica_sources)

C:\Users\chris\Anaconda2\lib\site-packages\mne\viz\ica.pyc in plot_ica_properties(ica, inst, picks, axes, dB, plot_std, topomap_args, image_args, psd_args, figsize, show)
    278         # --------
    279         # component topomap
--> 280         _plot_ica_topomap(ica, pick, show=False, axes=axes[0], **topomap_args)
    281 
    282         # image and erp

C:\Users\chris\Anaconda2\lib\site-packages\mne\viz\topomap.pyc in _plot_ica_topomap(ica, idx, ch_type, res, layout, vmin, vmax, cmap, colorbar, title, show, outlines, contours, image_interp, head_pos, axes, sensors)
    799     data = ica.get_components()[:, idx]
    800     data_picks, pos, merge_grads, names, _ = _prepare_topo_plot(
--> 801         ica, ch_type, layout)
    802     pos, outlines = _check_outlines(pos, outlines, head_pos)
    803     if outlines not in (None, 'head'):

C:\Users\chris\Anaconda2\lib\site-packages\mne\viz\topomap.pyc in _prepare_topo_plot(inst, ch_type, layout)
     69 
     70         if layout is None:
---> 71             pos = _find_topomap_coords(info, picks)
     72         else:
     73             names = [n.upper() for n in layout.names]

C:\Users\chris\Anaconda2\lib\site-packages\mne\channels\layout.pyc in _find_topomap_coords(info, picks, layout)
    613         pos = np.asarray(pos)
    614     else:
--> 615         pos = _auto_topomap_coords(info, picks)
    616 
    617     return pos

C:\Users\chris\Anaconda2\lib\site-packages\mne\channels\layout.pyc in _auto_topomap_coords(info, picks, ignore_overlap, to_sphere)
    671         # Get EEG digitization points
    672         if info['dig'] is None or len(info['dig']) == 0:
--> 673             raise RuntimeError('No digitization points found.')
    674 
    675         locs3d = np.array([point['r'] for point in info['dig']

RuntimeError: No digitization points found.
``

I followed pretty much the mne ICA tutorial and am stuck now. 
The Code is use is:
``
import numpy as np

import mne
from mne.datasets import sample
from mne.preprocessing import create_ecg_epochs, create_eog_epochs

# getting some data ready
data_path = 'C:\Users\chris\Desktop\S0.set'

from mne.preprocessing import ICA
from mne.preprocessing import create_eog_epochs, create_ecg_epochs

raw = mne.io.read_raw_eeglab(data_path)

# 1Hz high pass is often helpful for fitting ICA
raw.filter(1., 40., n_jobs=2, fir_design='firwin')

#work with eeg date, so eeg=True

picks_eeg = mne.pick_types(raw.info, meg=False, eeg=True, eog=False,
                           stim=False, exclude='bads')

n_components = 20  # if float, select n_components by explained variance of PCA

#set components equal to number of channels

method = 'extended-infomax'   # for comparison with EEGLAB try "extended-infomax" here
decim = 3  # we need sufficient statistics, not all time points -> saves time

# we will also set state of the random number generator - ICA is a
# non-deterministic algorithm, but we want to have the same decomposition
# and the same order of components each time this tutorial is run
random_state = 23

ica = ICA(n_components=n_components, method=method, random_state=random_state)
print(ica)

reject = dict(eeg=400e-6)

ica.fit(raw, picks=picks_eeg, decim=decim, reject=reject)
print(ica)

ica.plot_properties(raw, picks=0)

reject = dict(eeg=400e-6)
``

Any help would be very appreciated!

agramfort, 2017-12-15:
you need channel locations with montage to be able to plot topomaps.

see set_montage function.

mmagnuski, 2017-12-16:
BTW - ICA does not have any way to set montage. I'm not sure it is generally worth adding, but I've been multiple times in a situation when I forgot to add montage to a file, performed ICA and only then noticed I should add montage and perform ICA again - which takes much longer than changing montage in ICA instance.

larsoner, 2017-12-16:
Any object that carries around a `.info` could in principle have `set_montage`, so feel free to add the mixin to the ICA class to provide it.

Stoffander, 2017-12-29:
I did the set_montage and it worked perfectly, thanks a lot!

`` ##set montage for the channel locations

montage = mne.channels.read_montage(kind='filename', ch_names=None, path='datapath', unit='m', transform=False)
print(montage)

raw.set_montage(montage, set_dig=True)``

closed 2017-12-29 by Stoffander