mne-tools/mne-python #4837: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbe in position 13: invalid start byte

wangwuqi, 2017-12-16:
I failed to read a raw edf file, 

[12_0001_raw.zip](https://github.com/mne-tools/mne-python/files/1564572/12_0001_raw.zip)

```
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-3-40839df4f44a> in <module>()
     17 tmin, tmax = -1., 3.
     18 #raw=mne.io.read_raw_edf(r'C:\Users\wuqi\Documents\bci\BBCI\BCICIV_2a_gdf\A01T.gdf', preload=True)
---> 19 raw=mne.io.read_raw_edf(r'C:\Users\wuqi\Documents\emotive\DSI-24_Tools\DSI-Streamerv.1.08.09\12_0001_raw.edf', preload=True)

C:\Python36-32\lib\site-packages\mne\io\edf\edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
   1235     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
   1236                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
-> 1237                   exclude=exclude, preload=preload, verbose=verbose)

C:\Python36-32\lib\site-packages\mne\io\edf\edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

C:\Python36-32\lib\site-packages\mne\utils.py in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

C:\Python36-32\lib\site-packages\mne\io\edf\edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    162         input_fname = os.path.abspath(input_fname)
    163         info, edf_info = _get_info(input_fname, stim_channel, annot,
--> 164                                    annotmap, eog, misc, exclude, preload)
    165         logger.info('Created Raw.info structure...')
    166         _check_update_montage(info, montage)

C:\Python36-32\lib\site-packages\mne\io\edf\edf.py in _get_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    410     logger.info('%s file detected' % ext.upper())
    411     if ext in ('bdf', 'edf'):
--> 412         edf_info = _read_edf_header(fname, annot, annotmap, exclude)
    413     elif ext in ('gdf'):
    414         if annot is not None:

C:\Python36-32\lib\site-packages\mne\io\edf\edf.py in _read_edf_header(fname, annot, annotmap, exclude)
    594         # Recording ID
    595         meas_id = {}
--> 596         meas_id['recording_id'] = fid.read(80).decode().strip(' \x00')
    597 
    598         day, month, year = [int(x) for x in

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbe in position 13: invalid start byte
```

larsoner, 2017-12-16:
Can it be read by other software, such as `pyedflib` or non-Python readers?

agramfort, 2017-12-17:
how did you write this file?

wangwuqi, 2017-12-17:
I wrote this file with DSI-24, and this is the corresponding CSV file
[12_0001_csv_raw.zip](https://github.com/mne-tools/mne-python/files/1565358/12_0001_csv_raw.zip)
I think it might be the Chinese character causing the decoding error, but I can't change it.
[image]

agramfort, 2017-12-17:
yes it's the chinese character. Can you read this file with another EDF
reader?

wangwuqi, 2017-12-17:
No，failed with pyedflib too... 
```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-1-96f9ae11eaa4> in <module>()
      1 import pyedflib
      2 import numpy as np
----> 3 f = pyedflib.EdfReader(r'C:\Users\wuqi\Documents\12_0001_raw.edf')

pyedflib\_extensions\_pyedflib.pyx in pyedflib._extensions._pyedflib.CyEdfReader.__init__()

pyedflib\_extensions\_pyedflib.pyx in pyedflib._extensions._pyedflib.CyEdfReader.open()

pyedflib\_extensions\_pyedflib.pyx in pyedflib._extensions._pyedflib.CyEdfReader.check_open_ok()

OSError: the file is not EDF(+) or BDF(+) compliant (it contains format errors)
```

agramfort, 2017-12-17:
ok I guess the pb needs to be solved in the writer.

in the mean time you can hack the mne code to prevent the error.

wangwuqi, 2017-12-18:
I change the corresponding ASCII code in edf file, then it can be read

larsoner, 2017-12-18:
Closing for now, but if we see this problem on a lot of files we should `try:/except: warn` it. Having the date is typically not crucial to analyzing the data.

closed 2017-12-18 by larsoner

2220160084, 2019-09-19:
Hello, wangwuqi, i come accross the same problem, i cannot understand how did you change the ASCII code in edf file, would you please provide me some tips about how did you deal with this problem? thank you.