mne-tools/mne-python #4839: plot_compared_evokeds show_sensors & show_legend

athiede13, 2017-12-18:
Hello, this is the first time I am submitting an issue to git. I noticed that in the dev version, if I try to specify values (False or an int) for show_sensors & show_legend, they do not get taken into account. They just always get plotted in the bottom of the figure (legend on left and sensors on the right). Could this be checked?

Thank you!
Anja

mmagnuski, 2018-01-13:
Could you paste a smal code example that reproduces the problem?

athiede13, 2018-02-27:
Hey, sorry for the delay. Here is an example using the MNE sample data and code and the development version:

```
import os.path as op
import mne

data_path = mne.datasets.sample.data_path()
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True)
print(evoked)
evoked_l_aud = evoked[0]
evoked_r_aud = evoked[1]
evoked_l_vis = evoked[2]
evoked_r_vis = evoked[3]

conditions = ["Left Auditory", "Right Auditory", "Left visual", "Right visual"]
evoked_dict = dict()
for condition in conditions:
    evoked_dict[condition.replace(" ", "/")] = mne.read_evokeds(
        fname, baseline=(None, 0), proj=True, condition=condition)
print(evoked_dict)

colors = dict(Left="Crimson", Right="CornFlowerBlue")
linestyles = dict(Auditory='-', visual='--')
pick = evoked_dict["Left/Auditory"].ch_names.index('MEG 1811')
show_sensors=1
show_legend=2

mne.viz.plot_compare_evokeds(evoked_dict, picks=pick, colors=colors,
                             linestyles=linestyles, show_sensors=show_sensors, show_legend=show_legend)
```
It gives the following figure output: https://helsinkifi-my.sharepoint.com/:i:/g/personal/thiede_ad_helsinki_fi/EU4tyVpXmhROhE11EFSowBAB3cqHHNjMd7WQW13IkZ1RIg?e=KY4eUF

So, now the legend and sensors get plotted (and also they do not get plotted when their value is False), but the sensors should be in a different location (right top corner).

Thank you for checking this out!
Anja

agramfort, 2018-02-27:
what makes you think it's in the wrong place. If you do:

evoked[0].plot_sensors(show_names=True)

you see that 'MEG 1811' is on the center a bit left.

mmagnuski, 2018-02-27:
@agramfort I can reproduce, the topo legend should show in upper right corner of the figure with `show_sensors=1` but is plotted in the lower right instead. So `show_sensors=4` and `show_sensors=1` gives the same placement of the sensors legend. But for `show_sensors=4` lower-right is correct.

mmagnuski, 2018-02-27:
@agramfort I can replicate. With `show_sensors=1` the topography legend should be in the upper right of the figure, but it is located in the lower right. The issue can be pinpointed to [this line](https://github.com/mne-tools/mne-python/blob/857b1fed003004aed704db7ffaad10b2e94ac402/mne/viz/evoked.py#L1468) and the following one. The problem is how dictionaries work in the following case:
```python
dct = {'a': 2, True: 4}
dct.get(1, 1)
# returns 4
```

PR is here: #4976

mmagnuski, 2018-02-27:
Thanks for reporting @athiede13, please check if #4976 works for you.

athiede13, 2018-02-28:
Thanks @mmagnuski for the fast check! Newbie git question, how can I check it? The fix is not yet merged to the dev version, right?

mmagnuski, 2018-02-28:
@athiede13 Right, you can pull a PR branch this way:
```
git fetch origin pull/<PR_ID>/head:<BRANCHNAME>
```
where `<PR_ID>` is `4976` and `<BRANCHNAME>` is how you want to name the branch locally. Then you check that branch out (`git checkout <BRANCHNAME>`) and as long as you have an editable installation you should be ready to check if that works for you.

athiede13, 2018-02-28:
Thank you again @mmagnuski! It works now and shows the sensor and legend plots in the right locations for me (1 - right upper corner).

PR #4976 (2018-02-28): MRG: fix topo legend placement when `show_sensors=1`
fixes #4839, see my comments there.
closed 2018-02-28 by agramfort