mne-tools/mne-python #4841: Reset montage in raw?

cbrnr, 2017-12-19:
I was wondering if the montage can somehow be reset in an existing `raw` object. Suppose I did `raw.set_montage('biosemi64')`, but later I decide I want to reset the montage to nothing - how can I do that?

Does this make sense (during interactive data exploration it would be useful for me)? If yes, is it doable (I assume only if `update_ch_names=False`, because otherwise how should we reconstruct the original channel names)?

larsoner, 2017-12-19:
> but later I decide I want to reset the montage to nothing

You'll have to be more specific about what you mean. If you mean removing channel position information for all EEG channels, that's easy. If you mean properly resetting to a previous state of the object (e.g., keeping track of which EEG channels had positions set and names changed and only changing those), it will be much harder. In other words, doing `raw.set_montage(None)` that just wipes EEG channel locations is easy, doing `raw.unset_montage()` that should get me back the same `raw` object I had before doing `raw.set_montage('biosemi64'`)` is much harder and probably not worth it.

How is wiping channel location information useful during data exploration?

cbrnr, 2017-12-19:
It's maybe just a usability thing. If I can set a montage I'd expect that I can also unset it. I guess removing channel position information is sufficient for now (especially since names are not changed by default).

If this is not needed in MNE I can also just implement a custom function in MNELAB.

agramfort, 2017-12-19:
can you explain why you would want to reset it? just apply a new one if you
made a mistake no?

cbrnr, 2017-12-19:
@agramfort right, but you usually start out with no montage, so although a little OCD it would be nice to go back to a blank slate (e.g. if it turns out that there is no suitable montage file that works with the data or something).

larsoner, 2017-12-20:
+1 for `set_montage(None)` erasing EEG channel and fiducial positions

agramfort, 2017-12-20:
fine with me

cbrnr, 2017-12-20:
Please see #4846

PR #4846 (2017-12-21): [WIP] Remove montages with set_montage(None)
Fixes #4841.

Questions:

1. Do we care about the `set_dig` argument?
2. I'm not sure what the correct default value for `info["chs"][idx]["loc"]` is. `mne.create_info` fills this entry with `np.concatenate((np.zeros(3), np.eye(3).ravel())).astype(float)`, whereas `mne.io.read_raw_edf` seems to fill it with all zeros.
closed 2017-12-21 by larsoner