mne-tools/mne-python #4847: Can not import custom dataset

underscoredam, 2017-12-22:
I need to import a dataset for use in `mne-python`, but can not find a way to do so. The dataset was obtained from the paper referenced below. It is compatible with the MATLAB toolbox [bbci-public](https://github.com/bbci/bbci_public). In a nutshell, the dataset consists of MATLAB `.mat` files for continuous EEG signals, artifacts, montage, and event markers for multiple subjects. Please see the "Data Structure" on the link below for \[1\]. I can not find any way to import/process and use this data in `mne-python`, which is the only reason why I have not been able to use this awesome package.

Any help is appreciated. Thanks!

---
1. Jaeyoung Shin, Alexander von Lühmann, Benjamin Blankertz, Do-Won Kim, Han-Jeong Hwang and Klaus-Robert Müller, "Open Access Dataset for EEG+NIRS Single-Trial Classification," IEEE Trans. Neural Syst. Rehabil. Eng., online published. [Download dataset here.](http://doc.ml.tu-berlin.de/hBCI/contactthanks.php)

agramfort, 2017-12-22:
maybe this can help:

https://github.com/mne-tools/mne-python/pull/4019

but there is for sure some work to do.

underscoredam, 2017-12-22:
Thanks @agramfort, this surely helps! I've never used mne-python too much before; but it might be time to finally make a change for good. Meanwhile, should I leave this issue open, so that others will be able to discover it?

agramfort, 2017-12-22:
paste your solution when you have it and close then

alexandrebarachant, 2018-04-04:
This dataset is supported by the moabb : 
http://moabb.neurotechx.com/docs/generated/moabb.datasets.BBCIEEGfNIRS.html#moabb.datasets.BBCIEEGfNIRS

to load the data of subject 1 in the MNE format, just do

```python
from moabb.datasets import BBCIEEGfNIRS
dataset = BBCIEEGfNIRS()
data = dataset.get_data(subjects=[1])
```

larsoner, 2018-04-04:
Closing as it looks like there is a solution, thanks @alexandrebarachant 

BTW @alexandrebarachant if this is some package / pipeline you want to advertise in MNE feel free to open a PR to `doc/manual/datasets.rst`

closed 2018-04-04 by larsoner

alexandrebarachant, 2018-04-04:
@larsoner this is still a very young project so i will wait until thing stabilize to advertise here. We are aiming at ~20 dataset for the first release, all converted in MNE format :)

larsoner, 2018-04-04:
Great, feel free to open PRs to MNE to advertise / cross-link when you're ready

Hamidbme, 2019-05-28:
hi,
I use code but i have error 
code: from moabb.datasets import BBCIEEGfNIRS
dataset = BBCIEEGfNIRS()
data = dataset.get_data(subjects=[1])
error: https://pasteboard.co/IgNUTpK.png

Any help is appreciated. Thanks!

drammock, 2019-05-28:
@Hamidbme the code you've tried does not involve MNE-Python at all.  You should consider posting in https://github.com/NeuroTechX/moabb