mne-tools/mne-python #4849: Loading BDF files fails

youngmin91, 2017-12-26:
Traceback (most recent call last):
  File "<ipython-input-15-9239b716500e>", line 1, in <module>
    raw = mne.io.read_raw_edf(bdf_path, preload=True,stim_channel='auto')
  File "c:\users\ymna\mne-python\mne\io\edf\edf.py", line 1235, in read_raw_edf
    exclude=exclude, preload=preload, verbose=verbose)
  File "<string>", line 2, in __init__
  File "c:\users\ymna\mne-python\mne\utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "c:\users\ymna\mne-python\mne\io\edf\edf.py", line 176, in __init__
    last_samps=last_samps, orig_format='int', verbose=verbose)
  File "<string>", line 2, in __init__
  File "c:\users\ymna\mne-python\mne\utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "c:\users\ymna\mne-python\mne\io\base.py", line 371, in __init__
    self._preload_data(preload)
  File "<string>", line 2, in _preload_data
  File "c:\users\ymna\mne-python\mne\utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "c:\users\ymna\mne-python\mne\io\base.py", line 626, in _preload_data
    self._data = self._read_segment(data_buffer=data_buffer)
  File "c:\users\ymna\mne-python\mne\io\base.py", line 523, in _read_segment
    cals, mult)
  File "<string>", line 2, in _read_segment_file
  File "c:\users\ymna\mne-python\mne\utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "c:\users\ymna\mne-python\mne\io\edf\edf.py", line 252, in _read_segment_file
    fid.seek(start_offset + block_offset, 0)
IOError: [Errno 22] Invalid argument

I used mne.io.read_raw_edf(bdf_path,preload=True) function.
I got the error.
How can I fix it?

agramfort, 2017-12-26:
please share the file?

how was the file produced?

please report what you get with:

mne.sys_info()

youngmin91, 2017-12-26:
Thank you for replying

This is a information.

mne.sys_info()
results:
Platform:      Windows-7-6.1.7601-SP1
Python:        2.7.12 |Anaconda 4.2.0 (64-bit)| (default, Jun 29 2016,
11:07:13) [MSC v.1500 64 bit (AMD64)]
Executable:    C:\Program Files\Anaconda2\pythonw.exe
CPU:           Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 8 cores
Memory:        23.9 GB

mne:           0.15.dev0
numpy:         1.11.3 {lapack=mkl_rt, blas=mkl_rt}
scipy:         1.0.0
matplotlib:    1.5.1

sklearn:       0.19.1
nibabel:       2.1.0
mayavi:        4.5.0
pycuda:        Not found
skcuda:        Not found
pandas:        0.18.1
ᐧ

*------------------------------------------------------------------------*
*나영민 / Na, young min*
Dept. of Biomedical Engineering, Univ of Ulsan, Korea
Tel. 052) 259-1271
Mobile. 010-2051-9776
E-mail. nayoungmin39@gmail.com
*------------------------------------------------------------------------*

agramfort, 2017-12-26:
please try with more recent mne (last release is 0.15.2)
or with current dev master.

you did not tell me who wrote this file. With what other reader
can you read these files?

closed 2022-06-05 by sappelhoff