mne-tools/mne-python #4851: De-duplicate channel types

cbrnr, 2018-01-01:
As a follow-up to #4842, we should de-duplicate various definitions of valid channel types. Currently, these can be found in

- `mne.io.pick`: `get_channel_types`, `pick_types`, `_PICK_TYPES_DATA_DICT`, ... There are many more possibly relevant objects, but I need feedback which should be refactored and which should be left as is.
- `mne.io.meas_info`: `_kind_dict`

Anywhere else I forgot?

There should be only one place where valid channel types are defined. I suggest using `mne.io.pick` (currently it is inside `get_channel_types`, but this could be moved to a global dict in this module if desired.

agramfort, 2018-01-01:
I would also standardize stuff in pick.py

cbrnr, 2018-01-08:
@agramfort did you think of anything in particular in addition to the functions in `pick.py` I mentioned above? I think we should go through the whole module and choose what to refactor and what to leave alone.

agramfort, 2018-01-08:
did not have time to look yet. Start and we'll see how the diff looks like

agramfort, 2018-08-04:
@cbrnr I am not sure what you had in mind here? can you point to a code duplication or possible inconsistency?

cbrnr, 2018-08-06:
Currently the mapping between channel type and FIFF constants is implemented multiple times (at least in `mne.io.pick.get_channel_types` and `mne.io.meas_info._kind_dict`). There should be only one place where this  mapping is defined.

agramfort, 2018-08-06:
indeed !

+1 for using get_channel_types in meas_info and remove _kind_dict

larsoner, 2020-03-04:
We'll want to do this at the start of a new release cycle rather than right before release so I'm bumping to 0.21

commit 02db0b82 (2020-11-19): MRG, MAINT: deduplicate definition of FIFF constants (#8537)

* closes #4851

* fix tests

* modernize some tests along the way

* more flexible

* nest call within test
closed 2020-11-19 by larsoner