mne-tools/mne-python #4852: electrode locations cannot be found when plotting ICA components

aitorme, 2018-01-06:
Hi there,
I ran an ICA with more or less success following your steps, but when I try to plot the components I get the following RuntimeError:
`WARNING:root:Did not find any electrode locations the info, will attempt to use digitization points instead. However, if digitization points do not correspond to the EEG electrodes, this will lead to bad results. Please verify that the sensor locations in the plot are accurate.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/mne/preprocessing/ica.py", line 1425, in plot_components
    head_pos=head_pos, inst=inst)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/topomap.py", line 945, in plot_ica_components
    head_pos=head_pos, inst=inst)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/topomap.py", line 961, in plot_ica_components
    layout)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/topomap.py", line 71, in _prepare_topo_plot
    pos = _find_topomap_coords(info, picks)
  File "/usr/local/lib/python3.6/site-packages/mne/channels/layout.py", line 615, in _find_topomap_coords
    pos = _auto_topomap_coords(info, picks)
  File "/usr/local/lib/python3.6/site-packages/mne/channels/layout.py", line 673, in _auto_topomap_coords
    raise RuntimeError('No digitization points found.')
RuntimeError: No digitization points found.`

Although the electrodes are shown if I do `print(ica.info["ch_names"])`

agramfort, 2018-01-06:
see you set a montage file to specify channel locations?

aitorme, 2018-01-06:
I couldn't set any because I'm using an Enobio necbox and I didn't find any montage for this kind of EEG recorder, nor a way to create my own montage file. The channel locations are taken directly from the original `.set` file. Do I need to specify a montage for my data?

agramfort, 2018-01-07:
can you share the file? maybe the position of electrodes is not read
properly from your .set file.

aitorme, 2018-01-07:
sure, but I already read it in as a .fif when I run the ICA.
How can I share such a big file?

agramfort, 2018-01-07:
you should share the original file and the script you used to have the raw
object and the fif file.

you can use dropbox or google drive for example.

raw.plot_montage() should already tell you before ICA if channel locations
are well read.

aitorme, 2018-01-09:
I figured that importing one of the large montages available with many electrode positions would do the job, and now it seems to work fine, thanks!

mmagnuski, 2018-01-10:
@aitorme Alex meant `raw.plot_sensors()`.

mmagnuski, 2018-01-10:
oops, sorry, old Chrome session did not refresh and I was posting to the past state of this thread. :)

aitorme, 2018-01-10:
haha worry not, actually `raw.plot_montage()` never worked anyways, so your comment was useful

aitorme, 2018-01-16:
so apparently the ICA only works with a couple of participants, the vast majority of them crash. Since my eye channels are so bad, I was wondering if there is a way to implement a joint channel of two different channels (I would like to use an anterior and an eye channel as a joint channel for the ICA to detect the blinks better)

agramfort, 2018-01-16:
what do you mean it crashes?

aitorme, 2018-01-16:
python crashes and shuts down automatically

agramfort, 2018-01-16:
well this should not happen in Python. With MNE it is pure python code and
the program cannot segfault and crash.

you may have an installation problem ?

aitorme, 2018-01-16:
it also happens when I run it in Jupyter, should be independent of that?

aitorme, 2018-01-16:
I guess ram is not the issue since I have 16GB, but how can an install problem affect it and how could it be solved?

agramfort, 2018-01-16:
how did you install? are you using MKL with anaconda? what does
mne.sys_info() give you?

jasmainak, 2018-01-16:
did you monitor the memory usage with something like htop?

aitorme, 2018-01-17:
I do not use anaconda, although I have it installed, but I run it directly from the local/bin/

I did checked the memory usage with activity monitor (I use a mac) and the CPU as well and everything looked fine, no weird spikes

agramfort, 2018-01-17:
please try with anaconda with MKL and report back if it works.

I did not test recently but using pip install and mac os system Python
was before not super reliable.

aitorme, 2018-02-23:
Thanks for the help. I get the following error when running with anaconda 
`Did not find any digitization points of kind FIFFV_POINT_EEG (3) in the info`
after this warning
`WARNING:root:Did not find any electrode locations the info, will attempt to use digitization points instead. However, if digitization points do not correspond to the EEG electrodes, this will lead to bad results. Please verify that the sensor locations in the plot are accurate.`

aitorme, 2018-02-25:
I've seen that you also had this error when plotting sensors in issue #3120, do you know exactly where it comes from and how to solve it?

agramfort, 2018-02-25:
you need to have sensor locations in your file. Either with a montage or
reading them from the original file. If you have these in eeglab you should
be able to get them in MNE.

aitorme, 2018-02-25:
I do have the `.locs` file I get from eeglab, but it has an unconventional name that is not on the list of montages in the read_montage method. My system is a portable neuroelectrics necbox with 32 electrodes and I just cannot read it in with this method

agramfort, 2018-02-25:
you can make your own montage from numpy arrays if you have the locations.

let us know if you need more help

aitorme, 2018-02-25:
is the array just a vector with the channel names?
under which name should I pass it in order for read_montage not to throw an error?

agramfort, 2018-02-26:
you need to read the doc here:
https://mne-tools.github.io/stable/generated/mne.channels.Montage.html

and if you have not taken the time you should learn more about numpy.

For example reading this:
https://www.scipy-lectures.org/intro/numpy/index.html

jona-sassenhagen, 2018-03-31:
Can this be closed? @aitorme

larsoner, 2018-04-04:
Closing but feel free to reopen if still an issue

closed 2018-04-04 by larsoner

aitorme, 2018-04-05:
sure, I never managed to run the ICA anyways so I had to skip that bit

barlehmann, 2020-06-01:
I am not sure if this is related to the exact same issue, but I have been wrangling with problems around digitization points and sensor locations for EEG files for weeks. I have read through the sensor locations tutorial several times on the MNE website but have not yet been able to fix the issue. I am using .edf files from Emotiv Flex (32channel) as well as other 19-32 channel devices and their respective .edf files. I am able to plot the raw data without issue, but digitization points come out as "None". When I set montage using: montage_kind = "standard_1005"
montage =  mne.channels.make_standard_montage(montage_kind)
raw.set_montage(montage, match_case=False)
I get: ValueError: DigMontage is only a subset of info. There are 45 channel positions not present in the DigMontage. The required channels are: ['TIME_STAMP_s', 'TIME_STAMP_ms', '....'CQ_Cz', 'CQ_Fz', 'CQ_Fp1', '....] 
I have also tried the same with another file with the required channels being more typical channel names, but I continue to get this same error whenever I try to set the montage, plot topomap, or other functions of basic EEG analyses. It is likely that the .edf files I am working with do not have montage or digitization points associated with them, but the problem is that I cannot seem to set the montage either - whenever I try I get this above "ValueError: DigMontage is only a subset of info" (despite the fact that the channels are clearly visible in the channel names in the raw.info). Any suggestions would be greatly appreciated! (Please keep in mind I am a beginner in MNE, so simplicity will also be greatly appreciated.)

larsoner, 2020-06-01:
> raw.set_montage(montage, match_case=False)

On 0.20.5 you can do `raise_if_subset=False` and on 0.21 (`master`) you can do `on_missing='ignore'` to get past this error.

But it sounds like something more fundamental is wrong here. You need all EEG channels to have positions for anything topomap-related to plot properly, so using these `on_missing` is in general discouraged.

Channel types are an important concept in MNE, and it seems like you have channels that are not typed properly. For example, `'TIME_STAMP_s'` does not seem like an EEG channel, so it seems like it should be set to `'misc'` type using [raw.set_channel_types](https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types). Once you set all non-EEG channels to be non-EEG type (like `'misc'`), and you have a montage with all EEG channel names present, you should be able to do `raw.set_montage(complete_montage, match_case=False)` and things should work again.

@barlehmann where could we add to / where did you look in the tutorials where we could make this clearer? Maybe very early like:

https://mne.tools/dev/auto_tutorials/intro/plot_10_overview.html#loading-data

Setting channel types improperly can break / produce unexpected results in several steps, including Preprocessing (the next after "Loading Data"), so it seems like we should at least have a link there. @drammock WDYT?

larsoner, 2020-06-01:
Moving discussion to #7857