mne-tools/mne-python #4853: problem installing dependencies and MNE

yonlev, 2018-01-08:
i am attempting to install dependencies and MNE, but when i type pip install mayavi i get the following error message (see attached). 
any help is welcome,
thanks
Yoni

[image]

agramfort, 2018-01-08:
hi,

did you read:

http://martinos.org/mne/stable/install_mne_python.html

you recommend to install mayavi with conda and for python 3.6 use our
envrionment.yml file.

HTH

yonlev, 2018-01-08:
hi
yes of course, that is exactly what i followed though it recommends version 2.7. 
I currently have conda 4.3.30, and python 2.7.14. 
but installing mayavi on conda prompt gives me that error i mentioned

agramfort, 2018-01-08:
did you try to "pip install mayavi" or "conda install mayavi"

mmagnuski, 2018-01-09:
Judging from the first post and the screenshot - this is definitelly a result of pip install, conda should be fine.

yonlev, 2018-01-10:
thank you, yet even running conda install mayavi gives the following
[image]

mmagnuski, 2018-01-10:
This means that the requirements of these packages are in conflict (and conda is unable to solve that conflict). Your best solution might be creating a separate environment for mne+mayavi. You could try using the environment file mentioned in the mne-python install docs.

dafrose, 2018-01-19:
Hi and thanks a lot for this amazing package!

I just wanted to drop a note on installing mne on python 3.6, relating to what you wrote above:

> did you read:
> http://martinos.org/mne/stable/install_mne_python.html
> you recommend to install mayavi with conda and **for python 3.6 use our environment.yml** file.

We tried to install `mne` on python 3.6 using the above mentioned `environment.yml`. We exactly followed the recommendation in the (above mentioned) docs, saying:

> $ curl -O https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
> $ conda env create -f environment.yml
> $ source activate mne

but pip did something wrong and the install aborted, even leaving behind a broken `pip`. Instead we succeeded, by installing everything but `mne` using `conda` and the `conda-forge` channel and then installed mne using pip:

```
conda config --append channels conda-forge
conda create -n mne python=3.6 pysurfer scipy matplotlib scikit-learn mayavi jupyter spyder
pip install mne
```

We then verified the install using an [example in the docs](https://martinos.org/mne/stable/auto_examples/io/plot_objects_from_arrays.html).

Best,
DR

larsoner, 2018-01-19:
Maybe it's time to update the config file. If conda-forge has VTK for Python 3.6, then we don't need to use the `clinicalgraphics` repo anymore. @dafrose feel free to open a PR to update the `environment.yml` file and we can see if the CIs pass.

dafrose, 2018-01-19:
I've tried to move as much as possible in the `environment.yml` from `pip` to `conda`. Some issues that I encountered:
- `pysurfer` and `mayavi` conflict with `pyqt=5 ` &rarr; left them in pip section
- when trying to build mayavi with pip, I encounter the following error: `ModuleNotFoundError: No module named 'vtkIOParallelPython'` . I thought, this might be related to `VTK 8.1`, but fixing the version to `vtk=7` didn't help either

I forked at https://github.com/dafrose/mne-python

larsoner, 2018-01-19:
Can you open a PR to facilitate looking at the diff?

dafrose, 2018-01-20:
done! (PR #4885)

I had to remove the `pyqt` version requirement to be able to install `mayavi` and `pysurfer` with conda, since they require pyqt 4. Installing (and building) `mayavi` and `pysurfer` via pip failed on both a windows and a linux machine with `vtk` from conda-forge and `pyqt` v5.

I noticed that `mne` is already available on conda-forge for python 2.7 and 3.5 (in contrast to what the [docs](http://martinos.org/mne/stable/install_mne_python.html) suggest).  Wouldn't it be simpler and more reliable to suggest `conda install mne` as default installer?

agramfort, 2018-01-20:
I personally don't maintain the conda packages of mne or pysurfer (just pip)

if you can help us with this it's great

dafrose, 2018-01-20:
I don't have any experience with deployment (be it `conda` or `pip`) yet - but if you don't maintain the conda packages, then who does?

thx for maintaining on PyPi, though!

agramfort, 2018-01-20:
>
> then who does?
>
good people :) honestly I am not sure.

larsoner, 2018-01-20:
> had to remove the pyqt version requirement

This is bad because it forces matplotlib to version 1.5 (as 2+ requires
qt5). This is why we have pinned versions of pyface and so forth -- those
support qt5.

Thus maybe all we should change is where we get VTK (and maybe also how we
install MNE).

dafrose, 2018-01-20:
> Thus maybe all we should change is where we get VTK (and maybe also how we install MNE).

Seems so. I noticed that many of the pip-installed packages are available on conda-forge, so I moved them. Also building mayavi from pip failed on my (Windows) machine and I can't try my (Linux) office machine right now. 

I've just run a few test installs with python 3.5 and 3.6 . As it seems, mayavi installs alright from conda under python 3.5 with pyqt5 present but throws a conflict error (UnsatisfiableError) under python 3.6 (with pyqt5 present). Installing mayavi with pip under python 3.6 fails on my machine, so I can't get mayavi to run under python 3.6 and Qt5. The official docs and PyPI don't state any (!) python version dependency, so I don't quite understand the difference between 3.5 and 3.6 although Qt5 was present in both cases. Be it as it is. This is not an MNE but a mayavi problem.

jjnurminen, 2018-03-22:
FWIW, I just tried to install according to the docs (Python 2.7, conda 4.4.10), resulting in:

```
root@jupiter:/home/jussi# /opt/anaconda2/bin/conda install scipy matplotlib scikit-learn mayavi jupyter spyder
Solving environment: failed

UnsatisfiableError: The following specifications were found to be in conflict:
  - anaconda-navigator
  - astropy
  - mayavi
```

larsoner, 2018-03-22:
Can you try `environment.yml` from #5012?

larsoner, 2018-04-04:
Closing as this should be fixed with the `master` install instructions

closed 2018-04-04 by larsoner

ayoubbenaissa, 2018-11-16:
Check this out 
http://martinos.org/mne/stable/install_mne_python.html

it is very efficient