mne-tools/mne-python #4855: MNE-Python failing test_io_set from mne.io.eeglab.tests.test_eeglab

smgutstein, 2018-01-09:
I have just downloaded the most recent (?) version of mne-python

```
commit 79959786786311aedd20da272fb4de94a20f16ef
Author: Eric Larson <larson.eric.d@gmail.com>
Date:   Wed Dec 27 05:56:13 2017 -0500
```

but it is failing the following test:

```
>>> from mne.io.eeglab.tests.test_eeglab import test_io_set
>>> test_io_set()
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:627: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:627: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192, got 176
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:627: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/__init__.py:627: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192, got 176
  from ._ufuncs import *
/usr/local/lib/python2.7/dist-packages/scipy/special/_ellip_harm.py:7: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from ._ellip_harm_2 import _ellipsoid, _ellipsoid_norm
/usr/local/lib/python2.7/dist-packages/scipy/special/_ellip_harm.py:7: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192, got 176
  from ._ellip_harm_2 import _ellipsoid, _ellipsoid_norm
/usr/local/lib/python2.7/dist-packages/scipy/io/matlab/mio4.py:18: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from .mio_utils import squeeze_element, chars_to_strings
/usr/local/lib/python2.7/dist-packages/scipy/io/matlab/mio4.py:18: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192, got 176
  from .mio_utils import squeeze_element, chars_to_strings
/usr/local/lib/python2.7/dist-packages/scipy/io/matlab/mio5.py:98: RuntimeWarning: numpy.dtype size changed, may indicate binary incompatibility. Expected 96, got 88
  from .mio5_utils import VarReader5
/usr/local/lib/python2.7/dist-packages/scipy/io/matlab/mio5.py:98: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 192, got 176
  from .mio5_utils import VarReader5
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Reading /home/smgutstein/mne_data/MNE-testing-data/EEGLAB/test_raw.fdt
Reading 0 ... 30503  =      0.000 ...   238.305 secs...
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Reading /home/smgutstein/mne_data/MNE-testing-data/EEGLAB/test_raw.fdt
Reading 0 ... 30503  =      0.000 ...   238.305 secs...
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Reading /home/smgutstein/mne_data/MNE-testing-data/EEGLAB/test_raw.fdt
Writing /tmp/tmp_mne_tempdir_3X7w0m/test_raw.fif
Closing /tmp/tmp_mne_tempdir_3X7w0m/test_raw.fif [done]
Opening raw data file /tmp/tmp_mne_tempdir_3X7w0m/test_raw.fif...
    Range : 0 ... 30503 =      0.000 ...   238.305 secs
Ready.
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Data will be preloaded. preload=False or a string preload is not supported when the data is stored in the .set file
Events like the following will be dropped entirely: ['rt', 'square'], 2 in total
154/154 event codes could not be mapped to integers. Use the 'event_id' parameter to map such events manually.
As is, the trigger channel will consist entirely of zeros.
Data will be preloaded. preload=False or a string preload is not supported when the data is stored in the .set file
Writing /tmp/tmp_mne_tempdir_uwBxz0/test_raw.fif
Closing /tmp/tmp_mne_tempdir_uwBxz0/test_raw.fif [done]
Opening raw data file /tmp/tmp_mne_tempdir_uwBxz0/test_raw.fif...
    Range : 0 ... 30503 =      0.000 ...   238.305 secs
Ready.
Reading /home/smgutstein/mne_data/MNE-testing-data/EEGLAB/test_raw.fdt
Reading 0 ... 30503  =      0.000 ...   238.305 secs...
[... 53 lines cut ...]
Reading 0 ... 30503  =      0.000 ...   238.305 secs...
1 events found
Event IDs: [2]
No events found, returning empty stim channel ...
The following EEG sensors did not have a position specified in the selected montage: [u'FPz']. Their position has been left untouched.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/numpy/testing/decorators.py", line 147, in skipper_func
    return f(*args, **kwargs)
  File "mne/io/eeglab/tests/test_eeglab.py", line 180, in test_io_set
    raw = read_raw_eeglab(input_fname=one_chanpos_fname, preload=True)
  File "mne/io/eeglab/eeglab.py", line 194, in read_raw_eeglab
    verbose=verbose, uint16_codec=uint16_codec)
  File "<string>", line 2, in __init__
  File "mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "mne/io/eeglab/eeglab.py", line 340, in __init__
    info = _get_info(eeg, montage, eog=eog)
  File "mne/io/eeglab/eeglab.py", line 109, in _get_info
    update_ch_names=update_ch_names)
  File "mne/io/base.py", line 2517, in _check_update_montage
    % str(missing_positions))
KeyError: "The following positions are missing from the montage definitions: [u'FPz']. If those channels lack positions because they are EOG channels use the eog parameter."
```
How should I fix this?

larsoner, 2018-01-09:
I see this error, too, but I don't really get how it creeped into our code with CI testing (?!?).

Can you use [git bisect](http://webchick.net/node/99) to figure out which PR broke it?

larsoner, 2018-01-09:
I found the error, the positions were never checked properly and a recent PR made it fix those checks

smgutstein, 2018-01-09:
Cool. Thanks.

PR #4860 (2018-01-10): MRG: Fix loc
We should assign `np.nan` for unknown locations instead of `[0., 0., 0.]`. This PR moves us in that direction. I need to see what other internal checks this breaks, and fix them.

Closes #4855.
closed 2018-01-10 by larsoner