mne-tools/mne-python #4873: Skipping dense skin surface topology check

ktavabi, 2018-01-13:
For reasons I don't really understand nor can clearly spot in MRI data the high-density skin surface routine `make_scalp_surfaces` fails b/c the srf topology is botched and causes `_assert_complete_surface` to throw...  
>RuntimeError: Surface outer skin is not complete (sum of solid angles = 6.2829 * 4*PI instead).

I've tried `--force` and cutting off the neck but still hitting this more frequnetly than I'd like.
So I thought a PR to skip the skin surface topology check might be useful?

agramfort, 2018-01-13:
what code are you referring to?

this command
https://mne-tools.github.io/dev/generated/commands.html#mne-make-scalp-surfaces
?

where is the error raised?

agramfort, 2018-01-15:
yes we should be less strict. Can you change the python code to force the
non-crash?

ktavabi, 2018-01-16:
>Can you change the python code to force the non-crash?

That is the intent. Initial thinking was to ignore/skip `_assert_complete_surface` for the skin srf.  But you're right in that we should reduce the passing criterion. Right now the error is raised if `np.abs(tot_angle / (2 * np.pi) - 1.0) > 1e-5`. Where did `1e-5` come from? and what would be less strict?

agramfort, 2018-01-16:
if you use force option then make it a warning not an error

PR #4878 (2018-01-18): MRG: Allow skipping topology check
Closes #4873.

@ktavabi can you see if this fixes your problem? Run with `-i` or `--allow-incomplete`.
closed 2018-01-18 by agramfort