mne-tools/mne-python #4875: Q: why raw.pick_channels(temporal) in lcmv example?

kingjr, 2018-01-16:
In the `plot_lcmv_beamformer_volume` example, why is there a selection of left temporal channels before source reconstruction?

```python
left_temporal_channels = mne.read_selection('Left-temporal')
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=True, eog=True,
                       exclude='bads', selection=left_temporal_channels)
```
 I don't understand the rational of removing good channels before a source reconstruction.

cc @agramfort @britta-wstnr

britta-wstnr, 2018-01-16:
Good point - I remember a conversation about that from the GSoC, but cannot remember the reason for keeping it, @agramfort ?

dengemann, 2018-01-16:
It must have been carried over from a historical example where this was
done to make things look good. I remember that results used to look bad
when using all sensors. We should see if this is still the case.

britta-wstnr, 2018-01-16:
I can have a look at that ASAP - probably tomorrow.

agramfort, 2018-01-16:
the rationale is the following. Beamformers are know to fail when you have
very
correlated sources which is possible to binaural stimulations. Although it
works
ok on all sensors on sample I wanted to show how to restrict the LCMV to
the temporal
channels that only see the left auditory source?

shall we just clarify the doc?

kingjr, 2018-01-16:
IMO, examples should be restricted to a unique aim: here: how to do volume
source modeling. The issue of sync sources for beamformers belongs to a doc
/ tutorial, iif the results aren't affected.

agramfort, 2018-01-16:
unless you ahve time to develop this in a PR I would suggest to start with
a clear comment in this example.

britta-wstnr, 2018-01-17:
@agramfort actually, the correlated sources problem was most prominent when (for computational sake) the covariance matrix was calculated on the average of the trials. Now that we use epochs, we rarely run into that problem anymore - and if the results look good in this example, I would rather vote for showing that it works than using an old "hack" :smiley:

agramfort, 2018-01-17:
are you saying that LCMV with bilateral auditory stim works if you compute
the data cov on Epochs?

britta-wstnr, 2018-01-17:
most of the time it does, yes.

agramfort, 2018-01-17:
can we just add a note then?

britta-wstnr, 2018-01-17:
If I see it right, this example only uses the left-ear auditory stimulation trials anyway. Was the approach used to get the (weaker) ipsilateral activity then?

agramfort, 2018-01-17:
I don't think that was more to it than showing how it was possible to
select sensors to run LCMV...

sarangnemo, 2018-01-18:
I'd prefer not to propagate the myth that one can't see auditory activations with a beamformer without special tricks. The beamformer seems to perform excellently on this dataset using all the sensors, so why should we give people the impression otherwise?

agramfort, 2018-01-18:
you really think it's a myth? Maybe I've hanged out too much with MNE
people? :)

sarangnemo, 2018-01-18:
Perhaps better described as obsolete knowledge, and yes, clearly. :-) As @britta-wstnr pointed out above, it came up during the days of taking the covariance of the average, and even then it was only occasionally. We went through ~15 participants in our [2006 IEEE paper](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3066091/) before finding one that failed "enough" in order to demonstrate our workaround!

The reason is that the sources need to be very [highly temporally correlated (>0.85) ](http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=1159147) for it to become a problem, but in most people, the two auditory cortices aren't quite so synchronized, and have some delay between them.

Now, with covariance based on epochs (and optionally time-frequency approaches), it's even less likely that the correlations will approach that level. I suspect that moving from spherical head models to more accurate ones has also helped... I literally have not seen it come up since those days. In fact, I'll challenge anyone to contribute an auditory MEG dataset that fails on this pipeline! And of course we'll see what can be done about that, if somebody manages to break it. ;-)

agramfort, 2018-01-18:
ok fair enough then. Feel free to send a PR to remove this selection then

britta-wstnr, 2018-01-19:
I will have a look at it! PR follows.

larsoner, 2018-04-04:
Was fixed in #4887

closed 2018-04-04 by larsoner