mne-tools/mne-python #4876: Issue with raw.plot()

suntracks, 2018-01-16:
I am using a custom data format (*tdms from LabVIEW) . Used pandas to read data into MNE raw, however there seems to be some issue with the way the data is stored. 
I am seeing blobs of vertical lines. 
Current shape of numpy array - (8, 914585)

[image]

larsoner, 2018-01-16:
The scale factor is probably wrong. If these are EEG data they should be in units of volts, you probably have them in microvolts. Try changing the "scalings" parameter for your channel type (or multiply the channel values by e.g. 1e-6) to see if it fixes the problem.

suntracks, 2018-01-16:
Thanks @larsoner,
That solved the problem.

closed 2018-01-16 by suntracks