mne-tools/mne-python #4877: Bug: LCMV dipole orientation selection

davidhaslacher, 2018-01-17:
When the dipole orientation is computed in make_lcmv the weight matrix W is initialized as W = np.dot(G.T, Cm_inv) but never updated for each dipole index k as Wk is computed - thereby the method simply returns the weights for some canonical direction. For the "max-power" orientation, for example, it simply returns the weights for the first canonical direction: W = W[0::3].

agramfort, 2018-01-17:
@britta-wstnr can you have a look?

britta-wstnr, 2018-01-17:
Hi @davidhaslacher , please note that slicing of numpy arrays as done for ``Wk``  creates a view of the original array. This means that if you change ``Wk``, it will also change ``W`` (the data is "shared").

davidhaslacher, 2018-01-17:
Ooops, sorry! Thanks for the clarification.

agramfort, 2018-01-17:
close if it's not a bug

closed 2018-01-17 by davidhaslacher